EdgeTX/edgetx #5467: External nano modules missing on radios with standard JR bay

JimB40, 2024-08-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When selecting modules on radios with normal JR bay all modules using nano JR bay are missing. X9D, X9D2019, Jumper T16, RM TX16S etc
I've sold almost all JR bay modules leaving nano ones and using JR bay -> JR nano bay adapter.

While using full size JR modules on nano bay radios is clunky and inonvinient the other way, using nano module on radio with full JR bay using adapter very useful.

### Expected Behavior

Radios with JR bay should have JR nano bay modules enlisted.

### Steps To Reproduce

-

### Version

2.10.3

### Transmitter

RadioMaster TX16S / TX16SMK2

pfeerick, 2024-08-20:
AFAIK, there are only three modules that exist *only* in Nano form, and thus not selectable:
- XJT Lite
- R9M Lite
- R9M Pro

Are there any others? This is not a bug, but a very deliberate change => #3624 

Something to keep in mind is that unless you 1) do the external access mod AND 2) have a Nano to JR module adapter that supports that (which I don't think can even exist, as the extra connections needed on the Nano module format have nowhere to go on the JR Micro bay)... they will never work with a JR bay radio. 

Just FYI, there is no such thing as a JR Nano bay... JR Micro is the larger module format and Nano is the smaller one, and appears to have been created by FrSky, not Japan Remote Control.

All yours @3djc :hand_over_mouth: :rofl:

3djc, 2024-08-21:
Depending on radio module combination, some may not be displayed of radio doesn't have a full usart module (https://github.com/EdgeTX/edgetx/blob/005bd19588705bc76075a38fdaf1690a639efb14/radio/src/gui/gui_common.cpp#L897)

JimB40, 2024-08-21:
Okay followin naming.
1. JR Micro Bay (JRMB) - bigger module
2. Nano Bay (NB) - smaller module

I wasn't aware about differences between hardware implementation of those two

Here are NB modules I have
[image]

From Left to right.
0. BetaFPV JRMB -> NB adapter
1. TBS nano
2. Happymodel ELRS 2.4
3. BetaFPV ELRS 900
4. Ghost nano
5. FrSky R9M Lite PRO
6. FrSky XJT Lite
7. Multimodule nano (missing on photo)

AFAIR almost all worked via adapter (hasn't checked 3 yet)
As I have 2.10 on all radios and NB modules are disabled for JRMB radios can't make tests.

Anyway AFAIU 90% cases they will work with some exceptions.

3djc, 2024-08-21:
Crossfire (ELRS or TBS), Ghost, Multi don't have specific size entry, so they are CRSF, GHOST, MULTI, been JR or nano doesn't matter

R9M Lite Pro / XJT Lite:
- do require full UART that some radio don't have
- require small module to be declared. On radio with JR bay, that means adding " -DMODULE_SIZE_SML=YES" to your compile

pfeerick, 2024-08-21:
Well, here are only three nano modules that won't be shown on a JR bay radio, listed in my earlier comment 🤣

Another point to consider is that ACCESS JR Micro bay modules will be excluded from being shown if the radio doesn't have the hardware capabilities, or hadn't have the externalaccessmod. The same is true for Nano bay radios also - although I don't think for them we have any externalaccessmods documented.

JimB40, 2024-08-24:
> R9M Lite Pro / XJT Lite:
> 
> * do require full UART that some radio don't have
> * require small module to be declared. On radio with JR bay, that means adding " -DMODULE_SIZE_SML=YES" to your compile

Did wahyou wrote ie build 2.10.3 fot TX16S with  -DMODULE_SIZE_SML=YES
but the only one new modyle on list is R9MLite
no XJT Lite

3djc, 2024-08-24:
Dont think there is a full uart for the pxx2 on xjt lite, unless external access mod is done