EdgeTX/edgetx #5470: Radio settings and calibration should be separated

pfeerick, 2024-08-21:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

A

### Describe the solution you'd like

B

elecpower, 2024-08-21:
Refer #381

pagrey, 2024-08-21:
Does the voltage calibration belong in settings? Does the number and type of switches belong in settings? If you can't answer in a concrete way which thing is a setting and which is a calibration then maybe splitting them doesn't make sense.

The radio.yml file will always depend on the hardware configuration so why split it up? Seems like the UI in companion for managing the calibration settings is the main issue.

pfeerick, 2024-08-21:
Thanks Neil, was bugging me as I was sure I had created or seen an issue
for this before.

pfeerick, 2024-08-21:
> Does the voltage calibration belong in settings? Does the number and type of switches belong in settings? If you can't answer in a concrete way which thing is a setting and which is a calibration then maybe splitting them doesn't make sense.

It depends on what you mean by "settings". Settings UI? There is nothing wrong with the current UI. Settings file - it should be split, so that the radio settings is seperate to the hardware settings. i.e. consider the situation where you have two TX16S - you currently can't just copy the radio.yml between them... you need to calibrate afterwards. Wouldn't be necessary if you had a seperate `calib.yml` or `hardware.yml` file the RADIO folder. Or a EL18 and TX16S (which have completely different switch configurations)... again, not an issue if they are broken into a seperate file. 

> The radio.yml file will always depend on the hardware configuration so why split it up? Seems like the UI in companion for managing the calibration settings is the main issue.

No, it won't... how does the screen brightness, audio levels, or configured general functions (ok, I know this is the murkey middle ground) specifically depend on hardware? Note, the murkey middle ground could be resolved with virutal/assignable controls, which is itself another request around here. (i.e. rather than configuring your arm switch as SF, you configure it as ARM, and you then assign that on a per radio basis). 

Duplicate of #381

closed 2024-08-21 by pfeerick

elecpower, 2024-08-22:
> Thanks Neil, was bugging me as I was sure I had created or seen an issue
> for this before.

Forgiven as it was 5000 issues ago