EdgeTX/edgetx #5480: LUA playNumber with PREC2 - last decimal missing

NafetsRC, 2024-08-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

playNumber(123, 1, PREC2) says: „one point two volts“

### Expected Behavior

should say: „one point two three volts“ or ‚twentythree‘

### Steps To Reproduce

Put this line in a function LUA

same behavior in v2.9.4 AND v2.10.3

### Version

2.10.3

### Transmitter

RadioMaster TX16S / TX16SMK2

3djc, 2024-08-25:
The TTS engine is currently only generating audio with 1 digit after the coma. That dates OpenTX, and is designed to keep a balance between accuracy and annoyance of too long audio speech

NafetsRC, 2024-08-25:
Thanks 4 Info !

pfeerick, 2024-08-26:
So this is actually a bug that has been presentin OTX since 2.0.0?

https://github.com/opentx/opentx/blob/70867aaeb56c45ca2026df911d5b7434825e3888/radio/src/lua/api_general.cpp#L1111-L1124

We really should either fix that in code in our side (i.e. special handling if PREC2 param given), or remove

> * `PREC2` plays a number with two decimal places (for a number 123 it plays 1.23)

from the documentation and code ;)

3djc, 2024-08-26:
PREC is needed to as a divider for the value, but doesn’t drive how many digits will be said.. Yes it is a bug in the documentation

NafetsRC, 2024-08-26:
> We really should either fix that in code in our side (i.e. special handling if PREC2 param given), or remove
> 
> > * `PREC2` plays a number with two decimal places (for a number 123 it plays 1.23)
> 
> from the documentation and code ;)

Think, this would be really a great idea !!!

There are also some other people, who where misleaded from this…

cheers

closed 2026-08-31 by pfeerick