EdgeTX/edgetx #5481: Companion Sim still not OK lat , lon is output without a comma

koaxheli, 2024-08-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

When simulating the csv file, there is a space between lat and lon in the GPS column and is therefore not recognized as a comma is expected.

### Expected Behavior

In the old version 2.10.3 the space was replaced by a comma which has now been removed.

### Steps To Reproduce

Load a logged csv table and view the result 

### Version

2.10.3

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### Anything else?

Two pictures attached showing how the data from the GPS sensor is logged in the model and incorrectly analyzed in the Companion simulator.
[image]
[image]

pfeerick, 2024-08-26:
What version / build of Companion are you actually running now? As that is not a screenshot from 2.10.3 ;)

I am unable to reproduce the error with the comma with the [latest build of Companion for Win64](https://github.com/EdgeTX/edgetx/actions/runs/10558436236/artifacts/1854483014) (nor on the Linux version I tested this morning)... and I specifically tested for presence and absence of the comma, and it works fine for both with no error message.

[image]

koaxheli, 2024-08-26:
Version 2.11.0 latest feat(cpn): update telemetry simulation to include FrSky Hub (#5475)
The error occurs when I load a cvs log file and go to playback then the file is played back without a comma.

koaxheli, 2024-08-26:
Output of the simulation is all at GPS with zeros
[image]

nrw505, 2024-08-26:
Have you tried with the latest `main` branch? I don't believe a nightly build with the fix has happened yet.

koaxheli, 2024-08-26:
Sorry, I had the wrong version.
Now the right one and it works
chore(cpn): Stop SimulatedGPS refactoring, fix GPS latitude/longitude errors ... #2325

pfeerick, 2024-08-26:
Yeah, it was the next build (or any after) that you wanted for the GPS telemetry fix.  Glad to hear it's working for you now :)

closed 2024-08-26 by pfeerick