EdgeTX/edgetx #5487: Change the state of customizable buttons via special functions

TGS78, 2024-08-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I use one of the six customizable function buttons on my Jumper T20 to enable f.e. the Betaflight Air Extension. But I would also like to automatically switch off the Air Extensions when I change to angle mode. I know this function can be already achieved via the Betaflight Modes, but the LED of the customizable function button would then be incorrect. So changing the button state via a special/global function would be great. Additional Benefit: unused buttons could be also misused as f.e. a flashing indicator LED.

### Describe the solution you'd like

Add the possibility to SET and TOGGLE the state of customizable function buttons via special/global functions.
How to deal with grouped buttons would probably need additional discussion.

pfeerick, 2024-08-29:
I would be strongly against this. We don't allow special functions to change the state of any other switches, and I don't think this should be any different. At best, _maybe_ some additional logic added to customisable switches directly. But I'm not sure how that would look. 

On a personal note, I have air mode only configured on the FC - and is only enabled when not in STAB or HORZ mode... I see no need for it to be manually controlled :shrug: I have it off in STAB mode primarily as I use that for takeoff and landing. Although I do need to play with BF 4.6s EzLanding mode, if that works as well as advertised, I may just leave AIR permanently on.

3djc, 2024-08-29:
While I'm also strongly against it for physical switches where we would have a logical state not matching physical position, I'm actually in favor of that for customisable switches, because they have software controllable state indication, therefor the physical state (represented by led status) would keep matching logical state.

It would however still have to obey the switch setup: if you have a group of 3 switch that is always on for example, it would not allow to turn all off, or have 2 on. It would act as if you had pushed the button yourself, no less, but no more either.

On a side note, by putting your AIR and ANGLE in a group, you probably would achieve what you wanted

pfeerick, 2024-08-29:
Probably not, as he would have ANGLE/HORZ/ACRO in a group so they are
mutually exclusive, but AIR is another mode that can work in conjunction
with any of those three. You'd need nested groups for that to work lol.

I still do not beleive SF is the right mechanism, and that it should remain
in the domain of customisable switches.

3djc, 2024-08-29:
Let me give you other usage exemple, that definitely would need SF:

- petrol powerwered plane, you want to control choke, but that it doesn't stay on for more than a minute
- you want to control your craft landing gears, but also that it retract those above 60m

In those two scenario, a customisable switch with SF would be perfect, has it gives you visual feedback of the state, yet provide automation

pfeerick, 2024-08-29:
No, it would only need to be done via a SF *if* you make it that way. Or are you saying it can't be a conditional set/reset option in the customisation screens section? Just because it is there, it would not that you don't get access to things like logical switches... that would like saying that mixes can't make use of logical switches :laughing: Look back at what I said... I didn't say I am against programmatic control of the customisable switches... but of SF being be the mechanism to do so.

3djc, 2024-08-29:
So basicaly we need to recreate the SF screen, but under custom switch section ???? I'm failing to see the logic of that

pfeerick, 2024-08-29:
Why exactly do you need to recreate the SF screen under customisable switches (CS from now on)? Don't you simply want a set or reset condition, for each CS , or CS group, which is only available depending on some rules about the CS setup. So perhaps a set/reset choice for each, when it is not in a group, which is either OFF, or specifies the trigger. If it is in a group, you then need to specify which CS in the group is to be selected, and what the condition for it to be selected is, etc. 

One of the problems with it being under special functions, apart from it being yet another place to look to figure out how you configured things to work, rather than it all being in one place, is that you need to take into account whether a CS is a momentary, or part of a group that is supposed to be always on... and it makes no sense that this validation logic should be part of special functions?!

But before have any chance of figuring what the UI should/could look like... we first need to ascertain exactly **what** the requirements are... rather than just try to tack it onto existing structures... even if it is partly being processed as part of SF code, _if_ that is where it makes sense to actually handle updating the CS. 

For instance, off the top of my head I think these are the constraints:
- if a CS is a toggle, and not part of a group... it cannot be automated - user input _cannot_ be overridden
- if a CS is 2POS, part of a group, without always on requirement, you want to be able to set a condition for each CS for it's on state
- if a CS is 2POS, part of a group, without always on requirement, you want to be able to set a condition for each CS for it's off state
- if a CS is 2POS, part of a group, with always on requirement, since one CS **must** always remain on, _any_ attempt to turn them _all_ off must be ignored, meaning you can only set an on condition for each CS

Am I missing any states/conditions/requirements?

How do we deal those CS rules if it is configured via SF UI, and not part of the CS UI? Plus, it really seems poor UX design, if you make changes to a CS/CS group which invalidates one of those rules, since now you need to go to special functions and edit it there _when_ you realise it isn't working any more. And yes, I know this is possible to do in other parts of the UI, doesn't mean we should make the same mistakes here.

3djc, 2024-08-29:
"if a CS is a toggle, and not part of a group... it cannot be automated - user input cannot be overridden"

Thats probably where it would be most usefull to be able to automate, see my examples above. In my view, you would be very simple: The SF would look like:

Trigger: whatever you want
Function: Push custom switch
V1: switch number (1 to 6 in current radios)

Thats all you need. What would happen is exactly what would happen if you manually pushed it. All you do is ask the radio to push that button for you when this conditions are matched

TGS78, 2024-08-29:
"We don't allow special functions to change the state of any other switches, and I don't think this should be any different"

I agree, changing the status of physical switch would result in a mess, that's why I asked only for the customizable buttons.
Because you already can change the startup state an exemption from this rule would imo not be an issue.

The simulated button press via SF like 3djc suggested would be best, then you as developer do not have to think about how to deal with button groups in your code. And in case somebody needs to SET a state to ON or OFF (not toggle) you can still achive by reading the current state and push again or not.

3djc, 2024-08-29:
Maybe a simple V2 AND condition would come handy

pfeerick, 2024-08-29:
> "if a CS is a toggle, and not part of a group... it cannot be automated - user input cannot be overridden"
> 
> Thats probably where it would be most usefull to be able to automate, see my examples above. In my view, you would be very simple: The SF would look like:

Er... no, I think you forget JC - a toggle does does not latch, and can not be always on if it is not in a group, so is only active **when you are actively pressing it** :laughing: :facepalm: 

I give up... if you want to make it so people can end up with broken configurations, and want to deal with the support issues, whatever. :shrug:

3djc, 2024-08-29:
I cannot understand how simulating button push could end up in broken configuration. A 'sf' push on a toggle without group will have it up for a short duration, like if you pushed it (not push and maintain)

pfeerick, 2024-08-29:
And a SF off on a toggle that is on? Which can only be on while the user is
holding it down?

And a SF OFF to a cs that is a member of an always on group?

And a SF ON/OFF to a cs which is koe invalid as its configuration changed,
but the SF hasn't been updated as you need to edit things in multiple
places?

I am trying to consider *all* of the possibilities first, not *just* the
one use case of interest.

3djc, 2024-08-30:
I see where your concern is now, but I've not suggested you can set a CS on or off. As a reminder, the SF I'm proposing:

Trigger: whatever you want
Function: **Push** custom switch
V1: switch number (1 to 6 in current radios)
And I'm thinking about adding:
AND : and switch
Duration: time during which button will stay "pushed".

As for the SF been invalid, we have the same possibilities in so many case, reset a timer for example, everything around telem. Here it is actually better since the 6 buttons should still be there. If you push a button configured to none, then none will happen, it is not even invalid, just useless, but no different as if you would push said button with your finger

pfeerick, 2024-08-30:
> I see where your concern is now, but I've not suggested you can set a CS on or off. 

Neither have I... At no point have I said anything about CS configuration being changed by the SF (as that is something that would be for a Lua script to do, as part of mutel setup script, perhaps)... I'm have solely been taking about the input to a CS.

> As for the SF been invalid, we have the same possibilities in so many case, reset a timer for example, everything around telem.

Which is why I have being saying "we already know things can break there that way, is there not a better way?" Since normally we learn from mistakes, rather than continue to repeat them... Or so I'm told 🤪😂

3djc, 2024-08-30:
As said, there is no repeating mistakes as the 6 switches are there (and cannot be configured out), so how could it be invalid ?

Gerold68, 2024-08-30:
The topic "Change the state of customizable buttons via special functions" sound some kind of crazy dangerous to me.

Actually we were able to solve all of the needed control functions by the given possibility of SF and so on.
Changing the state of a Switch, Potentiometer should be avoided. This hardware gives haptic feedback to the user. 
This hardware actuators should the the starting action of a software procedure and not the other way around. 
It could be a shortcut of doing things with less effort, but the possibility of doing the required things is given.

Just my 5 cents.

3djc, 2024-08-30:
@Gerold68 do you own a radio with customisable switches (only Jumper does those currently) ? If you don’t, you probably can’t really grasp what those are, because as I said initially, I fully agree with you on physical switches, but those CS are in essence nearly a full software thing

Gerold68, 2024-08-30:
@3djc looking at my T15, you are right. So, if it is all about theses buttons I am with you. Maybe the missing haptic feedback is the reason that I do not use this buttons (also the ones of the TX16S) for things like flight-modes.
Tnx for your explanation. I didn't get it the first time. :-)

pfeerick, 2024-08-30:
> As said, there is no repeating mistakes as the 6 switches are there (and cannot be configured out), so how could it be invalid ?

Er... can you say that again? I have no idea at all what you mean there... and I don't think you meant to contradict ***yourself*** from one reply to the next :rofl: 

Sidenote... maybe add a feedback option - none, beep, haptic - to the CS page/section - which applies to all CS buttons?

3djc, 2024-08-30:
Yeah, was thinking the same about your sidenote.

What I was trying to say here (and the one earlier) is that the configuration of the custom switches is irrelevant for the SF I have in mind, The function is PUSH BUTTON, and the parameter is 1 to 6, matching the physical button. So the SF will never be invalid as the 6 physical buttons will always be there; no matter how CS are configured.

In practice, where we look "is button SW1 pressed", it would become "is button SW1 pressed OR is SF PUSH BUTTON 1 active"

pfeerick, 2024-08-30:
Ok, I'm with you now there... so because you are not setting or toggling the state of the CS, but simply mimicking user input of a physical press, all the existing CS logic safeguards will take care of things like the grouping, and always on state. 

And I presume  any attempt by the SF to toggle the button while the user is pressing it get rejected also? Or were you thinking of cheating and using the AND to avoid having to do that? :innocent: i.e. I'm now curious as a scenario where that condition would be used in the context of a SF like this...

3djc, 2024-08-30:
Well, if you push and the SF push at the same time, result is the same, it is pushed (so cheating with the AND, yeah).

Could you elaborate on your "I'm curious" statement, I'm sorry I do not understand what you mean there

pfeerick, 2024-08-30:
Yeah, I was just hoping it wasn't going to cause something like it being turned of (since in the case of an always on toggle, each press toggles it's state)... _while_ you are pressing it...

You said you were thinking of adding "AND switch"... so how would it be used in conjunction with this? i.e. you've already triggered this SF to be called, so why the AND SW? Use case? I'm assuming you have something in mind already?

3djc, 2024-08-30:
ok understood.

For the AND switch, i was thinking about checking current state (something like if this switch logical state is on, the push it to do a OFF only), but that would be subject to config changes. I could remove it, but then you would use a LS as a trigger, and that LS would be subject to config changes (but that config change dilemma almost cannot be solved by software only)

pfeerick, 2024-08-30:
I'll have to think about that AND switch a bit more... I think it may have a benefit... not sure yet. Whatever you do it is going to be subject to config changes... hence why I was saying the UI should be... you know... in the one place... rather than scattered across 3 or more screens... :rofl: i.e. you could quite easily reduce it down to just a single choice field for each CS row... as you would would really only need the trigger field - know the CS from the row, and if set to none, it's disabled. 

*But* even doing it that way, you may still need to make changes on the inputs/mixes/LS page if you were to say re-order the CS... so doing it via a SF means the makes the CS page/section doesn't get any more cluttered :shrug:

3djc, 2024-08-30:
Well you can have as many SF lines as you want pushing the same buttons not only one

pfeerick, 2024-08-30:
Indeed, especially if that CS directly controls a channel... i.e. if were say a beeper channel, or control that only moves when the button is held down... etc.

3djc, 2024-08-30:
Not yet

3djc, 2024-09-02:
Please have a look at #5497, this seems to be running ok in the testing i've done on radio. For now (and likely for said PR), I have limited to basic functionality (ie no AND or Duration)

pfeerick, 2024-09-03:
Most of the instances I can think of that would require the AND would probably involve using a LS as the trigger anyway, but if gonna leave duration out for now, I think it will be worth adding the AND still as a followup... i.e. you could have the trigger being a timer LS, but the button push only allowed if a "switch" in a given position... i.e. I think it may become useful for the "push CS" SF with multi-stage/chain logic, or being able to use a single LS to trigger multiple things, without needing secondary LS lines.

3djc, 2024-09-03:
AND is probably a pain as I don't want to change cfn struct.*

I think i I can manage the following duration behavior:
- set to 0 : CS is virtually pushed for as long as the switch condition is true
- set to a non zero duration : no matter how long the switch condition is true (shorter OR longer), the CS will be virtually pushed for the said duration.

So lets say your switch is a toggle switch, and you set duration to 2s
- briefly push your toggle, CS is pushed starting this toggle active, and stays push for 2s
- push and maintain your toggle for 3s: CS is pushed starting this toggle active, and stays push for 2s, meaning that after 2 secs, your switch will still be true for 1s, but the CS will not be pushed anymore

If that behaviors makes sense, I'll code that

pfeerick, 2024-09-03:
Aw... now you're just teasing me... AND but not AND? 🤣 

Yeah, that makes perfect sense :)

closed 2024-09-14 by pfeerick

s-ol, 2025-09-06:
This currently doesn't work if configured in `Global Functions`, either the option should be removed or fixed there.

3djc, 2025-09-06:
Will check it anyway

PR #5497 (2024-09-14): feat: special function to push customisable switches
Thanks @philmoz for colorlcd UI

Fixes #5487