EdgeTX/edgetx #5492: Several telemetry sensors do not work on Jumper T-Pro v2

Saradon85, 2024-09-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In firmware version 2.10.4, some telemetry sensors stopped working, including GPS and model battery voltage.

### Expected Behavior

Restoring the operation of telemetry sensors.

### Steps To Reproduce

Sensors do not appear detected.

### Version

2.10.4

### Transmitter

Jumper T-Pro V2

### Operating System (OS)

Windows

### Anything else?

Device Jumper T-Pro v2, eRLS 3.4.3, frequency 2.4GHz.
Betafpv nano receiver. Flight controller Omnibus F4 Pro v2, iNAV 7.1.1.
On the factory firmware of the transmitter, telemetry contained more than 20 sensors. After updating EdgeTX to the latest version, only 10 sensors are detected, among which there is no GPS and battery voltage.

pfeerick, 2024-09-01:
Have you done anything on the flight controller (iNav) side? As only 10 sensors (without knowing which) sounds more like telemetry has been turned off / not re-enabled on the FC side. i.e. you may only be getting the built in ELRS telemetry sensors, not any from the FC.

3djc, 2024-09-03:
What is your module baudrate ?

Trahht, 2024-09-04:
RadioMaster Pocket cũng vậy. Cảm biến gps và cảm biến hiển thị vĩ độ và kinh độ của nó đã mất.

Saradon85, 2024-09-13:
I apologize, I was away and couldn't check and respond. Today I connected iNAN and indeed I did not enable telemetry. It’s strange, in previous versions I didn’t enable it separately; telemetry was active by default.
Once again, I apologize for the trouble and thanks for the tips.

closed 2024-09-13 by Saradon85