EdgeTX/edgetx #5496: wrong servo pulses

HubertHuber, 2024-09-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I analyze the servo signal at the receiver output, I keep seeing pulses of the wrong length every few seconds. It's always just a single pulse. The length of the false pulse corresponds to the MIN or MAX value in the OUTPUT definitions.

### Expected Behavior

Without stick movement servo pulses should always remain constant.

### Steps To Reproduce

Hold throttle stick in front or rear position and analyze the corresponding servo signal at receiver pin.

### Version

2.10.4

### Transmitter

FrSky X9 Lite / Lite S

### Anything else?

[log.txt](https://github.com/user-attachments/files/16835760/log.txt)

3djc, 2024-09-02:
What receiver are you using? What you describe looks like the failsafe data sent on a regular basis

HubertHuber, 2024-09-02:
Receivers: Frsky Archer plus GR6, Archer GR6, G-RX6, RX6R, RX4R
I selected the wrong TX - it is not FrSky X9 Lite / Lite S but **FrSky Lite pro**
Failsave set to **HOLD** - definitly NO failsave situation
Please look at the attached logfile - we're talking about **JUST ONE FALSE PULSE** every few seconds

3djc, 2024-09-02:
Yes one every few seconds looks like failsafe data been sent (NOT the receiver being in failsafe). Could you try with failsafe set to receiver ?

HubertHuber, 2024-09-02:
failsafe set to receiver - **NO CHANGE**

HubertHuber, 2024-09-04:
Did some additional tests - problem exists under ELRS too.

HubertHuber, 2024-09-04:
Some more tests - **FALSE PULSES** at PPM-out (external bay) too.

3djc, 2024-09-04:
Here is a 1min log, not happening here

[putty.log](https://github.com/user-attachments/files/16873124/putty.log)

HubertHuber, 2024-09-04:
What is -102 in your log? Is it an SBUS value?

HubertHuber, 2024-09-04:
I checked the SBUS-out signal on my receiver - seems to be OK - NO FALSE PULSES

3djc, 2024-09-04:
Mine was pwm output measured with an esp32. If you get proper sbus out, it means the radio is doing its job.

HubertHuber, 2024-09-04:
I have to correct my SBUS-measurement. The time was apparently too short. The pause between false pulses is significantly longer with SBUS.
Another interesting finding is that the error does not occur when the SOURCE is a SWITCH.

3djc, 2024-09-04:
The thing is, what the radio sends is strictly the same, like there is no difference AT ALL, that you look at sbus receiver or pwm pulse. There is only one set of channels sent

HubertHuber, 2024-09-04:
NO FALSE PULSES when SOURCE is a SWITCH - measurement time more than 30 minutes.

3djc, 2024-09-04:
That leaves a question: either there is an edgetx adc issue reading the sticks (but then why only one of the axis, and none of the other axis / pots / sliders, and why only this radio when code is generic and used in all radio), or your hardware does have an issue

HubertHuber, 2024-09-04:
All 4 sticks and the 2 pots are affected. Do you suspect a problem reading the analog values?

HubertHuber, 2024-09-10:
An interesting finding:

If the parameter "Slow / Down" is set in Mixer, the false pulses disappear - that looks pretty strange ;-)

[image]

RDJay56, 2024-11-09:
Did this get resolved?  I'm seeing a similar thing happening.   My plane sitting on the ground I can see the flaps and elevator "twitch" down/up for just a millisecond.   Happens generally every 30-60 seconds.  TX is X-Lite Pro, Radiomaster ELRS module, RX is Radiomaster ER6GV.  Thanks.

HubertHuber, 2024-11-10:
No, there is no final solution, but I can live with my "WORKAROUND" - did you try it already?
Just set" Slow/Down=0.1" in all relevant mixers.
Regards HuHu

RDJay56, 2024-11-10:
Something odd going on it seems to me.  I went back to Edge 2.9.2 and the problem went away, however that messed with a bunch of my other programming.  Flashed back to 2.10.5 and added the .1 slow in all my mix lines.  Twitching didn't completely go away but seemed improved.

Thanks, Rj

RDJay56, 2024-11-11:
I've increased my "slow/down" all the way to .5 seconds and still see the glitches although not nearly as often.  Even tried adding "slow/up" (using both) and still have the glitches.

HubertHuber, 2024-11-11:
I'm currently on EdgeTX v2.10.5 - no change to v2.10.4
Trying to go back to EdgeTX v2.9.2 (your stable version) and see what happens.
Hoping "**3djc**" will have one more look at our Problem ;-)

3djc, 2024-11-11:
That won't happen since I don't have any archer gear or that radio

RDJay56, 2024-11-11:
I went back to EdgeTX V2.9.2.  Then used the 2.9.2 Companion to restore from a back up that was made with the same.  Glitches are gone.  Did to both my TX and 5 airplanes, tested all and is good.

BTW, I'm not using archer.  The TX's are Frsky X-Lite Pro using Radiomaster Nano modules and Radiomaster ER6GV RX's.

HubertHuber, 2024-11-11:
I also tried to go back to EdgeTX v2.9.2 but  could'nt find v2.9.2 Companion download.
Where did you download it?

RDJay56, 2024-11-11:
It was saved on my computer from when I was using it before.  I tend to keep old Companions, just install the new in a different folder.

pfeerick, 2024-11-11:
Companion for each release is always at the bottom under the Assets
heading, along with firmware.

https://github.com/EdgeTX/edgetx/releases/tag/v2.9.2

2.9.4 is the latest version for 2.9.x btw

https://github.com/EdgeTX/edgetx/releases/tag/v2.9.4

HubertHuber, 2024-11-11:
Thanks - unfortunately I always forget to open "ASSETS" to see the details.

HubertHuber, 2024-11-11:
@pfeerick
Hi Peter,
do you see a chance that this problem will be fixed in one of the next releases of EdgeTX?
Regards HuHu

pfeerick, 2024-11-11:
Can only be fixed if the cause is identified, and sounds like a very
strange issue at present. Which are sometimes the harder ones to pin down 😢

HubertHuber, 2024-11-12:
@pfeerick
Hi Peter,

_"can only be fixed if the cause is identified"_ - who will do that?

Is it enough that I opened this bug report (issue 5496) or do I need to do something else?

Regards HuHu

HubertHuber, 2024-11-12:
I just finished a test with EdgeTX v2.9.2
---> runtime more than 1 hour
---> more than half a million frames
---> **NOT A SINGLE GLITCH** (erroneous pulse)