EdgeTX/edgetx #5502: Adjusting GVar with trim switch work same as trim when held

acbates, 2024-09-03:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

You can assign adjusting GVars with trim switch, but pressing and holding will only change one step (Also a little annoying as the trim can still be active and you can hear the repeating/accelerating trim change sound as usual but the GVar only change one step).

### Describe the solution you'd like

To be able to hold the trim down, and have the GVar adjusted with each beep, just like a trim change. You can use GVars all over the place for great flexibility in model setup... but adjusting the values remains pretty clunky. Being able to hold a trip adjuster would be super handy.

Maybe it could be super handy to simply assign a GVar to a trim on the GVar screen, without having to set up a function for each side of the adjustment.

### Describe alternatives you've considered

There's no possible alternative to multiple increments trim-style adjustment outside of the trims themselves.

pagrey, 2024-09-03:
I'd use a special function:
Switch ON
Action Adjust GV1
Source Trm5
Enable
Now GV1 will always contain the value of Trm5, isn't this what you are asking about?

acbates, 2024-09-03:
> I'd use a special function...

I was using the function but only thought the increment +/- was the deal, had not known 'mixer source' worked in that way. Many thanks!!

closed 2024-09-03 by acbates

pagrey, 2024-09-03:
If you have anymore questions about EdgeTX there are people over at the [RC Groups thread on EdgeTX](https://www.rcgroups.com/forums/showthread.php?3916381-Official-EdgeTX-Discussion-Thread) that can help. It's probably a better place to ask questions. If you really find a feature that is needed, this is the place.