EdgeTX/edgetx #5504: Companion crashes when a SoarETX F5J model is active for 10min

neuhausj, 2024-09-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

1. The F5J model from SoarETX has a memory leak which leads to a "Lua disabled" message.
When restarting the simulator, the whole Companion app crashes.

This is repeatable on both Mac and Linux with latest build (2.10.4)

### Expected Behavior

Simulator should not crash Companion but restart "fresh".
SoarETX F5J model should be check for a potential memory leak.

### Steps To Reproduce

1. Start Companion and hit New button
2. Start the simulator
3. Create a new F5J model from SoarETX_v2
4. Wait ~10min, simulator will displayed "Lua disabled"
5. Click on restart simulator
6. Simulator+Companion will both crash

This is repeatable on both Mac and Linux with latest build (2.10.4)

### Version

2.10.4

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

macOS, Windows

### OS Version

MacOS 14.6.1 and Windows 11 23H2

### Anything else?

This is the Lua crash after ~10min:
[image]
Restarting the simulator will lead to a full Companion+Simulator crash

philmoz, 2024-09-03:
I can't reproduce this on my Mac - are you using the latest version of SoarETX_v2?

neuhausj, 2024-09-04:
I just double checked again, and used EdgeTX buddy to redownload the whole SD card content for the simulator.
I stopped it after 2000s (~30min) without having any Lua error.

I then compared the content of the SD card with my previous one.
Only diff was a personal template added (should make no ) + luac files for a few widgets.

On the brand new simulated SD, I then opened and closed all menus of the SoarETX widget and Lua got disabled after a few minutes.
I retried everything from scratch (redownload SD card content) and got the same result again after opening all menu from F5k SoarETX_v2 and letting the simulator run.
Here is the crash log and the SD card content (brand new from EdgeTX + only luac compiled for the F5k menus):

What is an effective way to debug memory usage in lua?

[logCrashSimu.txt](https://github.com/user-attachments/files/16870434/logCrashSimu.txt)
[SDSim3.zip](https://github.com/user-attachments/files/16870445/SDSim3.zip)

frankiearzu, 2024-09-14:
Will take a look at this.