EdgeTX/edgetx #5510: Logical Switch does not save / gets scrambled

TGS78, 2024-09-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I created a few logical switches with involved customizable buttons S1-S6.
The test at home was fine, but after I went to my fly spot nothing worked anymore, because some logical switch settings were suddenly messed up, f.e. S3 becomes ...ehm nothing and S6 becomes S16, see attached pictures.

It can reproduce this behaviour. The issue appears always when I turn the remote controller off and on.
Strangely enough, this happens only if the customizable buttons are the second AND/OR operant (V2), not if they are first (V1).
And the issue only appears if I use for example S3↓, but no problem with S3↑.
I have already cleared the logical switch, rebooted, and recreated it. The strange behaviour still exists.

[image]
[image]

### Expected Behavior

---

### Steps To Reproduce

---

### Version

2.10.4

### Transmitter

Jumper T20 V2

philmoz, 2024-09-05:
I can't reproduce this on my T20V2.

Can you upload your MODELS and RADIO folders (in a ZIP file).

TGS78, 2024-09-05:
affected was L15, L16
Meanwhile created an additional L14 with same settings, this also gets scrambled.
And now it becomes even more strange: created a L40 with same settings, but this works.

[T20V2_LS_CS_Issue.zip](https://github.com/user-attachments/files/16886389/T20V2_LS_CS_Issue.zip)

philmoz, 2024-09-06:
Thanks, I was able to reproduce with your file. Should be fixed by PR 5515.

PR #5515 (2024-09-08): fix(radio): CS values may be read incorrectly from yaml
Fixes #5510 

Incorrect length passed when converting int value for CS position.
If additional character is a digit then bad index value is calculated.

Also needs to be applied to 2.10.5.
closed 2024-09-08 by pfeerick