EdgeTX/edgetx #5513: Make Slow Up/Slow Down in Mixes adjustable by a global variable or pot/trim

Divefire, 2024-09-05:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Trying to dial in my MT12 for surface racing (off-road buggy) and I'd like to be able to make fine adjustments to the speed up/speed down parameters while driving. 

I've made mixes which means slow up and slow down for each side of the axis works in the increasing value but is ignored in the decreasing value (so the servo on the steering reacts at a reduced speed starting a turn either left or right but ignores it when returning to center or adding counter steer). 

But because the value is a fixed percentage or seconds value it's not practical to make adjustments during a race by going into menus. 

### Describe the solution you'd like

Would it be possible to add a global variable option to the list or to use trim/pots to adjust the value? 

### Describe alternatives you've considered

Alternatively, if anyone has a suggestion on how I can accomplish this adjustment using current options I'd be happy to hear them.

pagrey, 2024-09-06:
See issue #5118

Since no other surface radios have this feature my guess is it'll probably take a back seat for awhile.

philmoz, 2024-09-10:
Duplicate of #4832

closed 2024-09-10 by pfeerick