EdgeTX/edgetx #5514: TX16S, EdgeTX 2.10.x, no RC Link using "External RF" JR Bay (external module) and PPM\SBUS

packetkiller, 2024-09-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Using TX16S with RFD TXMOD v2 in JR port to a RFD 900x v2 to a Matek H743-WING. TX16S running 2.10.2 or 2.10.4. Flight Controller running Ardupilot 4.5.4 (Plane). Radio receives telemetry from FC and telemetry is viewable in Mission Planner via Mavlink from TXMOD WiFi. Telemetry is NOT visible in TX16S. RC link is not transmitted from TX16S to TXMOD (and thus never shows up on air side). Validated radios on both sides TX/RX sending and recieving data using scope. RFD radios are using default settings except for NETID custom value that is set to same on both.

### Expected Behavior

RC link sent to air side and then to FC. Telemetry from FC sent to TXMOD, then to TX16S, then discoverable in EdgeTX and live stream via telemetry LUA widget.

### Steps To Reproduce

Upgrade working firmware and SD card contents of EdgeTX from 2.9.x to 2.10.x.

### Version

2.10.4

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

This issue seems to be a root cause for issue #5064 (telemetry only issue).

pfeerick, 2024-09-06:
> RC link is not transmitted from TX16S to TXMOD 

This sounds somewhat fishy... shouldn't it be configured simply as PPM output? So if you have PPM output on the top pin of the module bay... you should at least have the control link, thus meaning that whatever the core issue is here, it appears that most of the reports in 5064 are common to the RFD TXMOD (v2 or modified v1), and changes around PPM/telemetry. Which would probably be that it defaults to No telemetry (as PPM doesn't have telemetry), and an option to allow MLink compatible telemetry... we probably need another option to add SPort telemetry...

packetkiller, 2024-09-06:
Even with the TX16S EdgeTX "External RF" option configured with PPM, MLink Telemetry, Channel Range CH1 to CH16, PPM frame set to 38.5ms, 300us, and -, there is no RC link but there is telemetry coming back to the TXMOD and viewable via the TXMOD WiFi in Mission Planner, but not via the TX16S EdgeTX (no telemetry discoverable). Powerup sequence is model, then TX16S, then connect to TXMOD via laptop WiFi.

mha1, 2024-09-07:
TX16s External RF set to PPM/No Telemetry or PPM/MLink, CH1..CH16, 38.5ms, 300us. -. A perfect 16ch PPM stream at the JR Bay CPPM pin. You've gotta find your problem at the receiving end of the PPM stream.

[image]

packetkiller, 2024-09-08:
> TX16s External RF set to PPM/No Telemetry or PPM/MLink, CH1..CH16, 38.5ms, 300us. -. A perfect 16ch PPM stream at the JR Bay CPPM pin. You've gotta find your problem at the receiving end of the PPM [...]

I agree. I will check each leg of the radios starting at the TX16S and working my way to the FC. The first test at the TX16S JR bay on PPM pin shows good PPM signal coming out. When I move the sticks, I see the signal move. Will status each leg to show others how I troubleshoot this...maybe it will help someone else. Next checking the 900x side to see if it got the PPM signal.

packetkiller, 2024-09-09:
So first leg across the radios shows no signal on pin 15 as expected. So the radios are talking as I can configure both from the TXMOD. For this forum, this means that the TX16S is working and spitting out PPM to the radio but the radio is not relaying that to the other side or the other side is not picking it up. I will keep working this with 3DR and report back when I have something to report.

packetkiller, 2024-09-09:
Update, managed to get ppm across link by resetting both radios to default, then setting S16:GPI1_1R/CIN = 1 on the TXMOD radio. Though I had this setting in place before and it did not work, this time, I did not run the "Go to First Run Wizard!". I have confirmed that the wizard sets something that breaks a standard PPM passthrough link. I am now going to attempt to validate on a flight controller and not just the oscilloscope.

packetkiller, 2024-09-09:
Update, am not able to get the flight controller to recognize the PPM signal. Going to try and move from PPM to SBUS and try again. RFDesign support has validated the PPM configuration. I have asked for config changes required to enable SBUS. Will see if this helps.

packetkiller, 2024-09-10:
Update, have now narrowed the issue to the radio software. Once a pair of radios are linked and configured, using the TXMOD web interface to change the configuration seems to "get stuck". I have been able to use the USB interface to change the radio settings but even that can get stuck and take a few tries to make a single change. Once changed, the pair do not communicate, thus the issue. Will continue to work with RFD to sort it out.

pfeerick, 2024-09-10:
Thank you for the updates, good to know making progressnarrowing the culprit, whatever it may be

packetkiller, 2024-09-11:
Finally figured it out. The main issue was related to the nature of how the radios save data to memory...meaning you must follow strict procedure for certain configuration changes. In short, when in doubt, just start back from all defaults and reconfigure fresh. It is a bit late here so will add another comment with the configuration details in the morning.

mha1, 2024-09-11:
Good to know. I was certain you'd find the problem at the receiving end of the PPM stream. Don't forget to close this issue.