EdgeTX/edgetx #5520: Curves screen cross hair gets stuck and erratic when changes are made.

Kevltan, 2024-09-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I edit the curves screen (a pitch curve in this case) the cross hair gets stuck when moving the related channel in my case the throttle stick. I cannot find a way to put this right. The inputs screen graph works as expected and so does the channel monitor screen.
It seems there is a problem here as I believe as the curve is edited the cross hair should follow the new path created.
With a 2pt basic line the cross hair with full stick deflection I get -100 -100 to 0,0 0,0 in the centre of the curves screen and moves no further and the cross hair freezes in the middle (and sometimesabove middle and jumps back) reading 0,0 unless I move stick to left and down where it returns to -100, -100.

### Expected Behavior

When editing a curve the cross hair should immediatley follow this new path that is created. Reseting model, powering off, retracing move in inputs screen does not rectify the problem. Surely the cross hair should mimmick the edits made in real time?

### Steps To Reproduce

1. edit a current curve in curve screen
2. move stick
3. cross hair gets stuck and jumps
4. inaccurate display of stick position top left hand corner of graphical representation.
5. No obvious way to resolve as above

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X12, RadioMaster TX16S / TX16SMK2

philmoz, 2024-09-08:
Not reproducible on my TX16S or T15.

Curves are not linked to specific sticks.
When you edit a curve (from the Curves page) the graph follows the position of the last stick moved. If you move the throttle the curve will follow the throttle position until you move another stick.

If you assign a curve to an input, then long press the curve name button to open the curve editor, then the graph will only follow the stick assigned to the input channel.

Kevltan, 2024-09-09:
https://github.com/EdgeTX/edgetx/issues/5520#:~:text=If%20you%20assign%20a%20curve%20to%20an%20input%2C%20then%20long%20press%20the%20curve%20name%20button%20to%20open%20the%20curve%20editor%2C%20then%20the%20graph%20will%20only%20follow%20the%20stick%20assigned%20to%20the%20input%20channel.
works

Kevltan, 2024-09-09:
@philmoz
If you assign a curve to an input, then long press the curve name button to open the curve editor, then the graph will only follow the stick assigned to the input channel.

Cross hair works on the curves screen it takes you into from the INPUTS screen, when long pressing the curve name, but then returningout of this and then going back into the CURVES screen it doesnt work properly again! Cross hair gets stuck in various positions. Try it and see what you get when a curve is edited and let me know your findings are, as mine is definately inconsistant and seems to be related the CURVES screen page.
Does this on both X12S and TX16S!

Kevltan, 2024-09-09:
@philmoz 
Please have a look at this video as you can see the cross hair traces as it should do in INPUTS page following your above instrutions, but when you go back to CURVES page the cross hairs are all over the place again!
https://github.com/user-attachments/assets/eaaa0a19-4a2d-4006-a7c7-67177d1ab9ac

philmoz, 2024-09-09:
Does this happen on all models? Does it happen on a new model with no special functions, logical switches and simple one line inputs and mixes?

If you can reproduce it on TX16S then please upload your MODELS, SCRIPTS, RADIO and WIDGETS folders from TX16S in a ZIP file and indicate which model(s) show the problem.

Kevltan, 2024-09-09:
[TX16SFiles.zip](https://github.com/user-attachments/files/16936969/TX16SFiles.zip)

philmoz, 2024-09-10:
I was able to reproduce with your files, thanks. Fix has been added to PR #5524

Kevltan, 2024-09-10:
@philmoz 
Thanks Phil, will try later when get home 🙂👍👌

Kevltan, 2024-09-10:
@philmoz
Thanks Phil, now works as expected. Thanks for the fast response as usual 👍😊👍😊👍😊

PR #5524 (2024-09-11): feat(color): use curve select button on outputs edit page to match inputs and mixes
As discussed in #4914 

- Use popup menu to select curve.
- Allow long press on curve button to jump to curve edit page.

Fixes #5520
closed 2024-09-11 by pfeerick