EdgeTX/edgetx #5521: Logical switch with functions "a>b", "a<b" etc. do not work correctly with GVs

RC-SOAR, 2024-09-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

LS's with functions of type "a<b" or "a>b" do not handle GVAR operands correctly. 
The value of the GVAR is divided by 10 before the comparison

### Expected Behavior

The value of the GV should be used, without being divided.

### Steps To Reproduce

1. Set GV1 = -80
2. Create a logical switch L1, function = 'a>b' 
3. In the Logical switch, set V1=Thr, V2=GV1
4. Observe the state of L1 while moving the throttle stick, the transition occurs when the throttle is at -8 instead of -80.

### Version

2.10.4

### Transmitter

FrSky X9D

### Operating System (OS)

Windows

### OS Version

Windows 10

philmoz, 2024-09-09:
The range for a GV is -1024 to 1024.
If you want to trigger the 'a>b' LS at -80% throttle then set the GV value to -819 (-1024 * 80 / 100).

RC-SOAR, 2024-09-09:
Hmmm.... it also fails if the range of the GV is altered to -100 to +100 in the GV properties.  Also, using function 'a > x' with GV1 as the first operand works as expected (straight comparison with constant value).

Unfortunately the workaround - supplying GV*10.24 - doesn't work for me as the value of GV must be preserved for use in another context. A workaround which _does_ work is to set the output of a channel to GV1 and do the comparison on the channel.

RC-SOAR, 2024-09-10:
Further thought, rather than the GV being handled incorrectly, could the channel value be the culprit? Its native range is -1024 to +1024, in comparison ops with GVs and constants it should be converted to  a percentage. Perhaps this is not happening with a>b etc functions.