EdgeTX/edgetx #5527: Option to automatically switch to a specific model when USB joystick mode is selected after you plug in USB.

Sea-Kerman, 2024-09-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Sometimes I forget to switch to my "joystick" model that has the radio modules turned off and some specific switch configurations, when using my TX for a sim.

### Describe the solution you'd like

In the Radio Setup menu, have an option for USB Joystick Model, where your options are your various models, "current model", and "select", with it defaulting to "current model". "Select" would bring up a model selection screen, in case you have multiple different joystick models.

### Describe alternatives you've considered

Perhaps the usb mode selection dialog could prompt you to check your models, or warn you if the current model has the radio modules turned on...

3djc, 2024-09-11:
I don't think it would be safe to actually switch model. Displaying model selection could however be an option

pfeerick, 2024-09-11:
That wouldn't really be an issue... you would only have to display a confirmation prompt first - i.e. after you plug in, and select USB joystick (or even if it is the default option)... "Switch to preset model '[NAME-OF-MODEL]'? [YES] [NO]" type dialog. And perhaps do trainer mode / telemetry checks first, and show a popup if unsafe to do that rather than the confirmation prompt. This would all be contingent on a new option being change from the default of "use current model"...

3djc, 2024-09-11:
yeah, thats what I meant, you need to kind of validation, not change model because you plugged a powerbank because radio was complaining of low voltage and you plugged top instead of bottom..