EdgeTX/edgetx #5528: Bump EdgeTX internal precision

3djc, 2024-09-11:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

bla

### Describe the solution you'd like

bla

### Describe alternatives you've considered

bla

pfeerick, 2024-09-11:
Well, they do say bigger is better, don't they? 🤭 

Personally, for us mere mortals... I would prefer leaving the display as is... as I'm not sure having say -32000 - 0 - 32000 is going to be more ... *meaningful* ... although I do then also understand it is not a true and accurate representation of the data being manipulated... so 😖 ... maybe it does need to be whatever resolution it ends up... probably 12bit until some new tech comes along...

3djc, 2024-09-11:
Leaving it at current scale does have the great side benefit of leaving current model programming untouched.

It does have the drawback to not add precision at programing (LS, SF,..), but I don't think it is needed there anyway

pagrey, 2024-09-17:
If it's ever done, maybe internal at first and then a display option if people ask? If the precision is displayed maybe an option that allows %, µs, 1K or 32K so people can choose and the radio can convert display values? As far as usage, with a scroll wheel 32,000 could be allot of scrolling, maybe too much to be useful at all on some radios.