EdgeTX/edgetx #5532: Enhancement for use of "Spezial Function Trainer"

ChrisOhara57, 2024-09-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When doing training with new pilots I set a logical switch that gives the control back to the master when a stick on the master radio is slightly moved (referring to source stick).
I think that's a very common method and it works fine as long as I use the slave signals in mixers.

But I cannot use the Special Function "Trainer" with this method.
It seems that the function simply replaces the values of the master Sticks with the values of the TR.. signals.
So when the student moves his sticks the control is immediately back at the master radio.
If the master has his sticks in neutral position the control then goes back to the student again.
And so I end up in an endless loop.

### Describe the solution you'd like

Proposal:
Can we have four additional sources to use with the logical switches.
These sources should always represent the sticks of the master radio (and never be swapped to the TR.. values of the slave radio).

### Additional context

At the moment I still use OpenTX.

froqstar, 2024-09-17:
Related discussion: https://github.com/EdgeTX/edgetx/discussions/3162

ChrisOhara57, 2024-09-17:
> Related discussion: #3162

good point; indeed (I haven'd found that)

As mentioned in #3162 a change in the existing software has its challenges.
That's why I proposed do install four new sources instead of altering the existing ones.
This would minimize the risks/efforts of the software modification.

ChrisOhara57, 2024-09-17:
I've left a description of all my Workarounds in issue #4916