EdgeTX/edgetx #5533: Flightmode UI issue with fade in/out on bw 128x64 screen

agnauck, 2024-09-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When a fade in/fade our value is set in a flight mode there is a UI issue on bw 128x64 screen. It looks like its trying to display the fade in/ fade out values over the trim values.

See FM2 and FM3 on this screenshot
[image]

[image]

### Expected Behavior

It should display the trim values only. There is no space for the fade/in fade out values on the 128x64 screen.

On the 212x64 screen its working fine, because there is enough width.
see:
[image]

### Steps To Reproduce

1. create a new flight mode
2. add fade in and fade out to the flight mode
3. look at the flight modes page

### Version

2.10.4

### Transmitter

FrSky X-Lite / S / Pro, RadioMaster Boxer

pfeerick, 2024-09-14:
Since there isn't enough space to show more, on bw128 a single character is drawn on the far right - `*` if you have both fade in and out, otherwise it will be a `I` for fade **I**N, or a `O` for fade **O**UT... so we might have to sacrifice a letter of the FM name to move the whole trim block over one character so it can be shown without overlapping the last trim when there are 4 trims... since if there are less than 4 there will be blank space, and if there are more than 4 but less than 8 there will be blank space on the second line.

agnauck, 2024-09-14:
yes, trimming the flightmode name by 1 character would be a solution. It will fit then and show properly as designed instead of overlapping the last trim.

To free up more space you could also consider to replace FMx with the name when present.
This is how its done on other pages like inputs or outputs. The first column with FMx is redundant and takes up 4 chars of the limited screen.

philmoz, 2024-09-14:
No need to lose a character from the name - there is enough space if we squash things up a bit.
[image]

philmoz, 2024-09-14:
Another option - remove the 'FM' before the number. This also makes room if using a trim as the switch:
[image]

agnauck, 2024-09-14:
@philmoz option 2 would be my favorite.

pfeerick, 2024-09-14:
Same. `!` switches are the worst case scenario here, and even they should be legible there.

PR #5536 (2024-09-15): fix(bw): adjust 128x64 B&W FlightModes tab layout to better fit values.
Fixes #5533
closed 2024-09-15 by pfeerick