EdgeTX/edgetx #5539: Trainer modes set with Special Function switch reversed

fhpage, 2024-09-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Running EdgeTX v2.10.2.
Appears to me that the Trainer modes are switched.
If I set up a special function:
Function: Trainer
Value: Axis
then trainer input only works when using a source of TR1-TR16.
If I change the Value to "Chans" trainer input works for the 4 axis without any additional mixes.

I am testing using external IRangeX for wireless buddy box. DSM_RX

### Expected Behavior

Mode Axis provide trainer input without mix entries.
Mode chans provide trainer input via T1-T16

### Steps To Reproduce

Special Function switch: Trainer mode Axis or Chans

### Version

2.10.2

### Transmitter

RadioMaster TX16S / TX16SMK2

pagrey, 2024-09-17:
It works as expected for me with a cable and using the iRX4 module with DSM_RX. Have you checked the channels monitor to be sure the slave radio is always outputting all the channels you want? Are the outputs working correctly on the master radio with their own controls on the channel monitor? On the slave radio do you have the channel range set to CH1-16 or whatever range you want to send?

fhpage, 2024-09-17:
It works for you with a cable?  This is a wireless configuration. 
What does "works as expected" mean for you?
With:
   Radio Settings, Trainer: each stick(or input) is assigned "Add"
   Special Functions set to "Trainer", "Chans"

Trainer radio is a Spektrum DXs
When the Special Functions is active. Channel Monitor shows seven channels Outputs active. No Mix. 
Master radio stick movement is not recognized! There should be because of the Radio Settings mention above.
If I add a mix line to one channel with a source of TR2, Channel Monitor shows both Mix and Output. This should not happen!
If I change the Special Function  to "Trainer", "Axis"; the only channel that shows any output is the channel that I added a mix line with source, "TR2". Both "Outputs" and "Mixes" show output. This is the expected behavior for "Chans".
Is this the expected behavior?

pagrey, 2024-09-17:
Works as expected means axis operates channels 1-4 only and channels in my case also operates channel 5. I tested wired and wireless just for fun. 

Both radios should operate all channels normally in the channel monitor with no special mixes. One special function to activate trainer mode. I'd forget about trainer for a minute and get the outputs for each radio operating normally first. Once you have both radios operating normally in the channel monitor then activate trainer and see if it works as expected.

fhpage, 2024-09-17:
The Axis variable should ONLY activate channels 1-4! If the trainer radio is a 7 channel radio I don't want channels 5-7 added to the mix! This is another bug...
What radio and version of EdgeTX are you testing with?
Both the Master and Trainer radios are working.
*My point is that "Mix" and "Chans" are reversed...*
I checked EdgeTX 2.10.0 on an X10S, wired. It does not work as expected.

pagrey, 2024-09-17:
Mix has nothing to do with this, you are causing problems with your mix. Your mix can override the way trainer works. Your mix should have nothing to do with any trainer function until you get channles and axis working.

It might help to create a new blank model on each radio so you understand the process. All you need to do is add a single input for a switch on channel 5 or line 5 on the input page of the new models, only the input page. Then setup the trainer on both radios. Be sure to leave input, mix and output tabs alone after you add the channel 5 switch. At this point your test channel 5 will work on channels but not on axis.

I'm using a Radiomaster Boxer with an iRX4+ and a Jumper T-lite both running 2.10.4, the current release version.

3djc, 2024-09-17:
https://manual.edgetx.org/color-radios/model-settings/special-functions
https://manual.edgetx.org/color-radios/radio-settings/trainer

fhpage, 2024-09-18:
Once again - apologies for taking up your time to respond. I did a complete test with a fresh program. **No issues.** One warning to others. Mixing the use of T1-T16 with a Special Function switch activating Trainer may produce unwanted results.

closed 2024-09-18 by fhpage

pagrey, 2024-09-18:
Glad you got it worked out. You'll also find help on the RC Groups [thread](https://www.rcgroups.com/forums/showthread.php?3916381-Official-EdgeTX-Discussion-Thread) about EdgeTX.