EdgeTX/edgetx #5542: Quick search in long lists via page key(s)

froqstar, 2024-09-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Especially with large sound packs it is very cumbersome to scroll through potentially 100s of tracks to find the right one. 
So a "fast scroll" functionality would be awesome to have. 

### Describe the solution you'd like

Could we use the "page" button(s) of the radio to jump to the entries starting with the next/previous alphabet letter when having a scrollable list open? 

### Describe alternatives you've considered

Would a press-and-scroll with the scroll wheel work as well? 
For Color UI maybe a second list "column" with all the first letters present in the list (or just all of them?) could be helpful as well, similar to how sources can be browsed by category.

philmoz, 2024-09-16:
2.11 file selector:
[image]

froqstar, 2024-09-16:
Nice! 
But for B&W it's still the same long list even in 2.11, correct?

3djc, 2024-09-16:
B&W don't enough memory to load long list

3djc, 2024-09-16:
That assumes that files are sorted in the directory, not sure they are necessarily

3djc, 2024-09-16:
They might be because you copied them all at once and they were sorted on pc. What happens if you add new ones ?

froqstar, 2024-09-16:
Okay, reading up on it with my very limited C++ skills I now see what the problem is. Thanks for that. 
So we can't assume that the files will be sorted in 100% of cases so we can't do something fancy and obvious like on color UI because it might not work in all cases. 

Still in most cases they should be, because people usually copy full sound packs or SD card contents onto their cards, at which point they are sorted until the order is changed manually (e.g. adding single files later).
So it might be doable on a best-effort basis if that is acceptable for the project - worst case the fast skip will become "unordered" at the end of the list and the user can still scroll through every list entry like right now, best case the search time for files can be cut to 1/10th or so. 

Thanks a lot for your patience.

3djc, 2024-09-16:
Function that work only in best case scenario don't look like a great idea to me

pfeerick, 2024-09-16:
Even if you copy entire sound pack at once, that is unfortunately no
guarantee... Windows likes to do things at random at times... Like copy the
last file first... But only sometimes... :-/

The only way I can see this working is if instead the folder was
specifically loaded one letter at a time "on demand"... rather than a
single scrollable list... but that will still chew into the memory
significantly in some cases, so I think this really isn't an viable option
for current generation b&w radios.

froqstar, 2024-09-16:
Thanks for the explanation

closed 2024-09-16 by froqstar