EdgeTX/edgetx #5543: cannot add model images and Theme only changes perimeter color...2.10.2 TX16s

jkoz29693, 2024-09-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

cannot add model images and Theme only changes perimeter color...2.10.2 TX16s. Used Buddy to update...did not work w/ 2.10.4 either

### Expected Behavior

Theme should fill entire background (works on my other TX16S; same config)
Should be able to select an image to go with Model info; cannot display image to Model. Images were copied from other working TX16S 

### Steps To Reproduce

See above; notice blue sky in 2nd pic, should be "Greenfields"

You can see problem unit is Carbon Fiber, second working unit is just plain black...
[image]
[image]
[image]

### Version

2.10.2

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

10

pfeerick, 2024-09-16:
Theme on your second radio is probably missing the wallpaper images for the theme, hence why they are not being shown. i.e. copy the THEME folder from your good radio to the other, to make sure all the files are there. Or install the themes from the themes directory - I'll add the link later. 

For the model images, I'm not sure... Have you doube checked the image for that model is present in the IMAGES folder, and is configured on that model?

jkoz29693, 2024-09-17:
thanks for your response

Not really certain what was causing my issues; I had done several updates w/o problems. 

I went back to bootloader (instead of Buddy; was first time I used) 

as well a I think I had a sketchy USB port on my PC 

I'm all set; we can close/delete this.
Thanks again!

closed 2024-09-17 by jkoz29693