EdgeTX/edgetx #5549: Jumper T14/T-Pro, T-Lite not logging when USB/VCP is connected, bug?

Johanl1964, 2024-09-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I connect USB/VCP (telemetry mirror) to Telemetry viewer there is no logging in the transmitter even though this is activated, when USB/VCP is not connected logging works as usual!

### Expected Behavior

Would like to be able to log in as usual even though USB/VCP is connected

### Steps To Reproduce

xx

### Version

Nightly (Please give date/commit below)

### Transmitter

Jumper T14

### Operating System (OS)

Windows

### OS Version

w10

### Anything else?

x

websquadpro, 2024-09-19:
try this

https://mega.nz/file/1vkGlJTL#yU-SJLlIilF0mx5qwpPOd3BM8iKZqOJLEQUStR76rrQ
If you don't have the c compliator, install it.(gcc or clang)

3djc, 2024-09-19:
Currently, sd logging is purposefully disabled when USB is plugged: https://github.com/EdgeTX/edgetx/blob/edb04cc83373f9f13dce8ad69e671343883dd75c/radio/src/logs.cpp#L253

I need to investigate if it is possible to remove that limitation

Johanl1964, 2024-09-19:
> Currently, sd logging is purposefully disabled when USB is plugged:
> 
> https://github.com/EdgeTX/edgetx/blob/edb04cc83373f9f13dce8ad69e671343883dd75c/radio/src/logs.cpp#L253
> 
> I need to investigate if it is possible to remove that limitation

Thanks, would be great if it can be solved! Even though you have an external USB device, there is still the need to be able to save the log on SD