EdgeTX/edgetx #5550: MT12 missing the "Telemetry" page in v2.10.X

duceppemo, 2024-09-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Using the factory firmware (MT12-edgetx.2.10.0-2023.10.30.bin), the "Telemetry" page is available (although the rest if the firmware is more buggy), but when flashing to the latest version ("Centurion" 2.10.4, also "Centurion" 2.10.0) via the Buddy, that 11th page is gone.

### Expected Behavior

I would expect the Telemetry page to still be available in all version of the firmware for the RadioMaster MT12 transmitter.

### Steps To Reproduce

1. Push "MDL"
2. Push "PAGE<" twice. -> no telemetry page

### Version

2.10.4

### Transmitter

RadioMaster MT12

### Operating System (OS)

Windows

### OS Version

Win 11 pro

pfeerick, 2024-09-20:
Only reason for the telemetry page to be missing is if you have it disabled in the settings... 

Under SYS -> Radio Settings, scroll down to Enabled features, and look for this option. It either needs to be ticked, or overridden on a per model basis:
[image]

For the per model setting, check under MDL -> Setup (scroll down to Enabled Features and expand it to see all the options). It needs to be GLOBAL to honor the Radio Settings level setting, or ON - else you have it turned off 😉):
[image]

duceppemo, 2024-09-20:
Thank you so much @pfeerick.
Your step-by-step guide was right on.
I guess I was the bug :sweat_smile:

closed 2024-09-20 by duceppemo

pfeerick, 2024-09-21:
No problem, I've done silimar myself 😂