EdgeTX/edgetx #5552: Edgetx-tx16s 2.7.1 not recognized by osx m3

MikeNats, 2024-09-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have a TX16S with Edgetx-tx16s 2.7.1 . When i plug it in to my MAC i get the option use it as joystick but my Mac is not recognize it. I have tried all the options (storage) etc and its impossible to read the device.

### Expected Behavior

When plugged in to my Mac book it should be able to read it

### Steps To Reproduce

1. open the TX16S
2. connect it to osx
3. select use as joystick
4. nothing is happenign

### Version

Other (Please specify below)

### Transmitter

Other (Please specify below)

### OS Version

Sonoma 14.4.1

gagarinlg, 2024-09-21:
Try another USB cable. Some cheap cables only have two wires for power, but no data wires.

You should consider updating your radio.

3djc, 2024-09-21:
Are you using usb-a to usb-c cable ? Because usb-c to usb-c cable don't work with tx16s.

MikeNats, 2024-09-21:
So I have upgraded to the 2.10.3. I am a dev and first time into controllers it was not easy please note that.  

I have used the cable provided by the controller and i used an Anker USB C Adapter (2 Pack),High-Speed Data Transfer did not worked. 

I have tried also other adapters too, and failed to connect. 

**What worked was to use my usb-c to usb-a cable from my mobile device with the Anker USB C Adapter.**

Thank you for your prompt response ! I would recommend that you include the right cable with the device and simplify firmware installation.

Thank you again!

closed 2024-09-21 by MikeNats

3djc, 2024-09-21:
We don't have any say with what manufacturers include with their radio. By the way, it works fine with the cable supplied by Radiomaster on my MacBook Pro, but I have to use a usb-c to usb-a active adapter

pfeerick, 2024-09-21:
For me, the cable that was supplied with the handset has always worked perfectly, as has any USB A to USB C cable (that is not a charge only cable).

Without knowing how you updated, and what you found hard about it, I think you will appreciate as a dev it will be hard ot "improve". I don't see how using https://buddy.edgetx.org/ and following the prompts is particularly difficult, especially given how it was significantly harder prior to that. But this is an open source project, as 3djc said, PRs to make improvements are always welcome.