EdgeTX/edgetx #5554: Support AUX2 Jumper T15

r1000ru, 2024-09-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

For connecting BLE Telemetry DIY and GPS, as variant

### Describe the solution you'd like

Add AUX2 in build action (Serial3 on PB10, PB11 and Serial6 on PG14, PG9), if there are no terrible obstacles.
[image]

### Describe alternatives you've considered

No alternatives

3djc, 2024-09-23:
Jumper have asked that this connector stay dedicated to something else.

mha1, 2024-09-23:
UART3 (the one close to the STM32) is already AUX1 and working. 3.3V on VCC though.

r1000ru, 2024-09-23:
> Jumper have asked that this connector stay dedicated to something else.

Maybe it's worth doing something else as one of the Serial ATA connection types (Telemetry Mirror/SBusIN/Jumper something else)?

druckgott, 2024-11-09:
The serial 3 is already working
https://github.com/EdgeTX/edgetx/pull/5445
I think also the other maybe already work because you can select two Aux. But i did Not test it.

r1000ru, 2025-02-19:
> Jumper have asked that this connector stay dedicated to something else.

Jumper hasn't released anything else. Besides, this something else should be supported by the firmware. Maybe open AUX 2 for all users? And when something else is released, add it to the list of usage options.