EdgeTX/edgetx #5555: [MT12] Integration into air radio

chofchop, 2024-09-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

As the first surface radio, the MT12 stands out from traditional air radios.

However, there are almost no benefits to this, and in fact, the disadvantages are noticeable, such as the UI becoming difficult to understand and companion support being delayed.

In the future, when new technology is introduced to mainstream air radio, it is expected that surface radio will be excluded.

### Describe the solution you'd like

I think MT12 (and surface radios that will appear in the future) should not be distinguished as "surface radio" but should be treated as part of conventional air radios.

[3djc](https://github.com/3djc) often says this.
"Edgetx for mt12 is fully featured, and everything you can do with air version can be done the very same way on surface version. Changes are mostly cosmetics around naming (like drive mode instead of flight mode)"
In other words, if change the name back to the original name, everything is resolved.
ST > Rud
TH > Thr
Drive mode > Flight mode ..... etc.

[3djc](https://github.com/3djc) also says:
"Changes are mostly cosmetics around naming (like drive mode instead of flight mode), **and work with 2 main axis instead of 4**"
RadioMaster has already given this answer. MT12 has an optional 2-axis joystick (or G-sensor).
X axis > Ail
Y axis > Ele
Now MT12 is completely equivalent to air radio. There are no disadvantages.

Also, surface radios are not limited to the gun grip + wheeler type. Traditional stick surface radios can be replaced with Mode 4, RTEA air radios.

Overall, I don't think EdgeTX needs to specialize in surface radio.

### Describe alternatives you've considered

I ported the firmware of TX12mk2 to MT12. Almost all I changed was hal.h.
[MTX12default.zip](https://github.com/user-attachments/files/17094680/MTX12default.zip)

It's an amateur job, but it's working well so far. I think this is a natural EdgeTX behavior.

3djc, 2024-09-23:
"UI becoming difficult to understand": UI is EXACTLY the same, no extra complication. It is only working change, and clearly, steering and throttle make much more sense than rudder and elevator to car pilots.

"In the future, when new technology is introduced to mainstream air radio, it is expected that surface radio will be excluded.": totally baseless, inaccurate statement.

Optional 2 axis sensor is already perfectly supported

chofchop, 2024-09-23:
> "In the future, when new technology is introduced to mainstream air radio, it is expected that surface radio will be excluded.": totally baseless, inaccurate statement.
Even companions don't support it yet. That's enough evidence.

3djc, 2024-09-23:
Companion PR is here: https://github.com/EdgeTX/edgetx/pull/5381, and from now on, surface radio will be supported as air radio, without any specific changes.

The code base for radio has been changed first to accommodate seamlessly both types, companion took a bit but to get there (and gaining further flexibility with that change).

Anybody familiar with EdgeTX development will confirm that you statement is simply plain wrong

chofchop, 2024-09-23:
> "UI becoming difficult to understand": UI is EXACTLY the same, no extra complication. It is only wording change, and clearly, steering and throttle make much more sense than rudder and elevator to car pilots.

That's why I say that "word changing" has made it difficult to understand.
I'm a "car pilot" and the throttle and rudder are easy enough to understand. And not rudder and elevator.

chofchop, 2024-09-23:
>Optional 2 axis sensor is already perfectly supported
> Companion PR is here: #5381, and from now on, surface radio will be supported as air radio, without any specific changes.

It looks like this PR doesn't even include any additional analog channels. It seems that it is still far from completion.

pfeerick, 2024-09-23:
> This is an opinion from the user side. It's not a question of right or
> Of course, I don't think everyone has the same opinion as me. However, it
> is also clear that my opinion is very reasonable.

> In the future, when new technology is introduced to mainstream air radio,
it is expected that surface radio will be excluded.

This is an invalid assumption, this is the just the first surface radio to
be incorporated, with the NB4+ to follow shortly. Now that the groundwork
has been done to support differences in surface and air focused radios,
which is **why** it is a "simple" hal change to support now.

And regarding Companion support, any new radio that is significantly
different to what has gone before has always lagged in Companion support...
The MT12 was a bigger challenge than most due to some underlying
assumptions in the core Companion architecture. And from what I can tell
from trying it out this afternoon, PR5381 appears to be both fully
functional and pretty much complete.

chofchop, 2024-09-23:
I understand that the development team wants to differentiate between surface radio and air radio. I hope that Surface Radio (MT12) will be fully functional soon.
On the other hand, the idea that it is better to integrate surface radio into air radio remains unchanged. This would greatly reduce the burden on the development team.

pfeerick, 2024-09-23:
That might have been the case say eight months ago, but not now since it
would involve *undoing* a lot of the work that has been done.

Surface focused radios are laid out differently to air focused radios, and
as time goes on will likely get some surface specific functionally added...
it is still early days. Keep in mind there is like 12 years of air-only
radio history in OTX/ETX here... so it will take a little time for surface
to get its own place.

inventor7777, 2024-09-25:
As an MT12 user, I don't feel left out at all. Like pfeerick said, there are some slight bumps in the road due to it being the first surface radio with EdgeTX (such as missing CP support), but as for the UI being less clean? I don’t think so IMO. I agree with both being as similar as possible though, other than the slight wording differences 3djc mentioned.