EdgeTX/edgetx #5560: companion program crashes when edit intern HF module at Pocket radio

meyer1111, 2024-09-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Companion program crashes when edit intern HF module (4in1, not CC2500!) at Pocket radio.

Internal RF 4in1 works very well. (ELRS has been replaced by 4in1 module)
But when i create a new model, the program crashes when edit intern HF module.
When i create a new model in the transmitter, the transmitter does not crash.
After this it can then be edited in Companion without any problems.
[image]

### Expected Behavior

Protocol selection without crash (4in1 internal HF-Module).

### Steps To Reproduce

1. create new model with companion
2. Protocol selection (internal HF module)

### Version

2.10.4

### Transmitter

RadioMaster Pocket

### Operating System (OS)

Windows

### OS Version

Win11 64bit, Win7 32bit

meyer1111, 2024-09-24:
Supplementary information: transmitter profile: language = de !!!
Maybe that's the problem?

Companion program crashes when edit intern HF module, regardless if CRFS or 4in1 module default setting.

pagrey, 2024-09-24:
It works for me. Be sure you change the radio profile settings to specify the internal module before you create the etx file. The module type is stored in the file. You can't open an existing etx file and change the internal module type. The setting only applies when the etx file is created.

elecpower, 2024-09-24:
I cannot reproduce in Win 10 64 bit.

elecpower, 2024-09-24:
> It works for me. Be sure you change the radio profile settings to specify the internal module before you create the etx file. The module type is stored in the file. You can't open an existing etx file and change the internal module type. The setting only applies when the etx file is created.

To clarify, in Companion you can change the radio profile Default Int. Module and each Models and Settings file Hardware Internal RF type at any time. 

The radio profile Default Int. Module is used to set the Hardware Internal RF type for New Models and Settings files. When you select New Model, the Hardware Internal RF determines which internal module types are available for the new model.

Changing the radio profile Default Int. Module will only apply to new Models and Settings files.

Changing the Hardware Internal RF of an existing Models and Setting file will trigger an update to the Internal RF of all models within that file. Should a model's Internal RF be incompatible with the new Hardware Internal RF then the model's Internal RF will be set to OFF.

elecpower, 2024-09-24:
@meyer1111 I would suggest you delete and recreate the Pocket radio profile and retry. There maybe some unrelated values that are triggering the crash.

elecpower, 2024-09-24:
@pagrey what version and radio profile are you using for your tests?

pagrey, 2024-09-24:
I compiled a version of 2.10.4 with the Pocket simulator module to test this issue, I haven't tried with any other radio just because I wanted to get a clean build to test this particular issue. It's running on Linux.

elecpower, 2024-09-24:
Thanks for the info. I'll recheck with this combo when I have some time.

meyer1111, 2024-09-26:
I delete and recreate the Pocket radio profile and retry.
With default setting: Multi.
Now: everything is fine! :-)

Thank you very much! :-)

[image]

closed 2024-09-26 by meyer1111

pagrey, 2024-09-26:
Great you got it all working.