EdgeTX/edgetx #5569: Touch screen not active when USB data cable removed after editing

Kevltan, 2024-09-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When editing SD card data on PC with USB cable connected and then removing, the touch sceen no longer works until a power reset. This is a long standing problem.

### Expected Behavior

Radio should function fully after removing USB cable.

### Steps To Reproduce

1. insert USB cable
2. edit
3. remove cable
4. no touch screen activity until a power reset.

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X12, RadioMaster TX16S / TX16SMK2

### Anything else?

Long standing problem

inventor7777, 2024-09-27:
I am not able to reproduce this on my end using a TX16s MKii and a MacBook Air.
I plug in the radio, choose USB storage, and it comes up on my computer. I added a file to the SD card and ejected the radio. The radio resumes functionality directly afterwards with no power cycle.

inventor7777, 2024-09-27:
(I’m just another EdgeTX user and thought I’d weigh in 😂)

Kevltan, 2024-09-27:
It doesn't do it every time and am trying to work out what sequence of events actually causes this. But there is definately an event that causes the touch screen input to stop working. A power reset always clears the problem, but is a bit of a pain!

.00000000000000000000000every time it does it

Kevltan, 2024-10-08:
This is only reproducable on my X12S transmitter NOT my TX16S. Usually happens after a SD card Dump to the PC.

Kevltan, 2024-10-16:
Has anybody had the chance to look at this or is this just due to the touch screen addition/modification to the X12S. Can this be fixed by firmware tweak?🤔

gagarinlg, 2024-10-16:
None of the developers have a X12S, so it is difficult to test and fix

pfeerick, 2025-02-07:
Has there been any change with the status of this with recently nightlies at all? I think unless someone gets their hands on a X12S, and does the mod, that we may have to put this issue aside as It is most likely an issue we can't resolve without suitable hardware (that none of us have).

Kevltan, 2025-02-07:
I can live with it. It just mean

pfeerick, 2025-02-09:
Thanks for the update, fingers cross it does get resolved at some point :)

closed 2025-02-09 by pfeerick