EdgeTX/edgetx #5570: Special functions get turned off when converting from OTX file.

Larry3215, 2024-09-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

I opened an OTX file and Companion 2.10.4 converted it to EdgeTx, but all the special functions got turned OFF. They were supposed to be ON.

### Expected Behavior

Special functions should have stayed ON.

### Steps To Reproduce

Opened E-Soar Plus from https://rc-soar.com/

### Version

2.10.4

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Windows 11 Home 23H2

elecpower, 2024-09-28:
Only selected OTX SF Actions have Enabled flag whereas as of 2.10 all ETX Actions do https://github.com/EdgeTX/edgetx/pull/3601
OTX SF Actions with enabled flag get converted to ETX with enabled on/off.
OTX SF Actions without an enabled flag get converted to ETX with enabled off.

The release notes for 2.10 https://github.com/EdgeTX/edgetx/releases/tag/v2.10.0 had this stated limitation:
"When updating, the following special and/or global functions may need to be re-enabled: Instant Trim, BG Music Pause, Vario, Screenshot, Racing Mode, Disable Touch, and Audio Amp Off"

RC-SOAR, 2024-10-01:
Just to add: the o/p raised this in relation to a setup in OTX 2.3 format targetting the X9D. This has a number of PlayTrack special functions which were disabled after conversion to ETX 2.10.4. 

It would be really nice if this could be fixed as all [my sailplane templates](https://rc-soar.com/edgetx/) are in OTX format (there are still a few OTX users out there, and the ability to import into EdgeTX means that only one master file needs to be maintained). 

Thanks for all the hard work, BTW, EdgeTX rocks!

3djc, 2024-10-01:
Otx is irrelevant here, he was just saying more (all) SF have enable field, the one where it was added need to be manually activated if desired. Coming from Otx or older Etx is the same

elecpower, 2024-10-06:
Fixed in 2.10.5

closed 2024-10-06 by elecpower

RC-SOAR, 2024-10-06:
Many thanks 😀

PR #5586 (2024-10-06): fix(cpn): set all sf/gf functions with repeat to enabled on otx import
Fixes #5570

Note: this is for 2.10.5 ONLY as binary import dropped in 2.11