EdgeTX/edgetx #5572: Lua: model.getTimer and model.setTimer are not updated to datastructure of 2.10.4

ChrisOhara57, 2024-09-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I'm using the lua functions model.getTimer and model.setTimer
The Datastructure of the timer have changed but it looks like the lua functions have not been adopted.

In the old (still lua) version the information of the switch was part of "mode".
Now it is a variable of its own.
But it is not possible to access it with the lua functions because they were not adopted to the new data structure.

### Expected Behavior

please adopt the lua functions (including documentation) to the new data structure.

### Steps To Reproduce

write a small script to set a timer with a hardware switch (like SA-).
You will see that the old syntax will not work anymore.
You will end ab with a timer that shows "off" and not Switch assigned.
And a call with ", switch=xx" in the parameter list will also not work.

### Version

2.10.4

### Transmitter

FrSky X9 Lite / Lite S

### Operating System (OS)

Windows

### Anything else?

[image]
[image]

ChrisOhara57, 2024-09-30:
Sorry, This is not a bug.

I've found out that I had a bug in my lua script.
This in combination with the out of date documentation made my wrongly believe that the the software is faulty.
Firmware seems to be ok, argument "switch" does work.
But documentation is our of date.

I do apologize, issue can be closed

lshems, 2025-11-28:
This is something to be fixed nonetheless in the documentation.

How to address that?