EdgeTX/edgetx #5573: Add file selector for LUA widget options

wimalopaan, 2024-09-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

LUA widgets can have options which select sources or values, but not files (on SD card)

### Describe the solution you'd like

A new option type `FILE` wich gives a drop-down list of files in a specific directory

philmoz, 2024-09-30:
The file selector popup is limited to a single directory (it does not support folder navigation) - so you would be limited to only being able to pick files from one place.
The folder to choose from would have to be fixed in the option definition, something like:
```Lua
  { "File", FILE, "/SOUNDS/en" },
```
which might be a problem if you want to allow users to select a sound from a different language for example.

wimalopaan, 2024-10-01:
I think, this limitation would be absolutely acceptable for the use cases in my mind: choose config files for LUA-widgets. The widget has a special folder for this.

philmoz, 2024-10-04:
Another question - string options are limited to max 8 chars for the user input value.
I don't think this would not be long enough for filenames; but increasing the length will increase RAM usage.

What is a reasonable size limit to cater for filenames?
Increase to 12 would cater for short file names (8.4) - is this enough?

gagarinlg, 2024-10-04:
> Another question - string options are limited to max 8 chars for the user input value. I don't think this would be long enough for filenames; but increasing the length will increase RAM usage.
> 

Is it worth it? People will complain about less free RAM on the older radios

wimalopaan, 2024-10-04:
Maybe I misunderstand something, but we are talking only about Widget-Options, not strings in general. So, B&W radios are not affected. And then also only those who use widget-options of the new FILE type.

philmoz, 2024-10-04:
> Maybe I misunderstand something, but we are talking only about Widget-Options, not strings in general. So, B&W radios are not affected. And then also only those who use widget-options of the new FILE type.

Yes it's color only so the extra RAM should not be a concern.
Unfortunately because of the way the firmware works the extra RAM is needed regardless of whether you use the widget option or not, everything is statically allocated. But we are only talking about few KB and color radios have 6MB+ free RAM.

PR #5605 (2024-10-18): feat(color): add FILE type for widget options
Resolves #5573 

Parameters:
- Option name
- FILE
- default value
- Folder to open to select file

Only the filename is returned for the option, excluding the path.
Filenames are limited to 12 characters maximum length.

> [!NOTE]
String widget option is increased from 8 characters to 12 characters

Example:
```C
  { "File", FILE, "testimg.png", "/IMAGES"},
```
closed 2024-10-18 by pfeerick