EdgeTX/edgetx #5574: Duplicate RX numbers don’t show a warning on MT12.

inventor7777, 2024-09-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On my MT12 (FlSky2A) if I use a duplicate RX number on a new model, it doesn’t show any warning or anything at all. 

### Expected Behavior

On my TX16s, when I try to use a duplicate RX number it pops up with a warning message underneath. I think something similar should work on the MT12, whether by haptic or a warning message.

### Steps To Reproduce

Bind to an RX with an RX number assigned. Try to create a new model with the same number and it won’t prevent you or show any kind of warning,

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster MT12

### Operating System (OS)

macOS

### Anything else?

I’m running today’s Nightly (9/30/24).

philmoz, 2024-09-30:
Color radios load basic info from all models into RAM at startup so it is possible to check if a receiver number is already in use.

The limited RAM on B&W radios makes this impractical, and  I don't think loading every model yaml file to check the receiver ID's when you create a model would work.

inventor7777, 2024-10-01:
@philmoz That makes sense. I hadn’t thought of that.

Maybe there could be a button instead such as “Check RX ID” and it could check the yaml, pop up and inform user if the ID is in use or not, then return to normal.

Just brainstorming here, i know that you’re pretty limited on B&W so I’m not pushing super hard for any fancy new features to clog up memory 😂

inventor7777, 2024-10-01:
Or maybe a txt file in the OS that stores all used RX numbers of the same protocol? Then the radio could load it without much RAM usage? 🤷‍♂️ (this would be actually pretty useful anyway even if there was no warning attached.)

3djc, 2024-10-01:
Or soon(tm) use companion that handles it also for bw radio