EdgeTX/edgetx #5575: yaapu script configuration menu not available

d-papworth, 2024-09-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Using two different Radiomaster TX12 MKIIs on ETX v2.10.4 and the yaapu script with Ardupilot, I cannot access the yaapu configuration menu to enable CRSF via the typical button presses of TELE and RTN.  
When I downgrade to ETX v2.9, it is possible to access the menu normally.  

This has been verified by myself multiple times as well as some others on the Ardupilot forums.  The only way I was able to enable the option was to downgrade to ETXv2.9, enable CRSF and then upgrade to v2.10 again.  An alternative method was shown to me in editing the yaapu config file directly from a PC to enable CRSF.

In my searching online, it seems others that were able to get yaapu working were likely working with ETXv2.9 based on the date stamps of their posts - since yaapu is kind of a niche case, I've not been able to find any others doing a fresh yaapu install on ETX v2.10 who may have had the same issue.

https://discuss.ardupilot.org/t/help-needed-with-yaapu-lua-script-and-telemetry-setup/113013/15?u=dpapworth
https://discuss.ardupilot.org/t/yaapu-passthrough-telemetry-on-elrs-zorro-setup-currently-not-working/86394/37?u=dpapworth

### Expected Behavior

When using the yaapu script - after entering the script main page in the model, I should be able to press Telem -> Return -> Telem to access a yappu config menu, where I can "enable CRSF" in order to allow the script to work with an ELRS RX on the aircraft.

### Steps To Reproduce

I have been able to test this on two separate RM TX12 MKII's and a Radiomaster Pocket, where the steps were as listed below and are referenced in the Ardupilot forum posts linked earlier.

Flash TX to ETX 2.10.4
Install yaapu script in model https://github.com/yaapu/FrskyTelemetryScript
Set yaapu to one of the DISPLAY Screens
Enter the main model screen
Press TELE
Yaapu script loads
No combination of TELE/RTN buttons will load the Yaapu Config menu
Flash TX To ETX v2.9
Enter main model screen
Press TELE
Yappu script loads
Press Telem -> Return -> Telem
Yaapu configuration menu loads
Scroll to "enable CRSF" 
RTN key
Script now loads telemetry info from aircraft
Update TX to ETX v2.10.4
Yaapu continues to function correctly, but not configuration menu is available 

(alternatively, "enable CRSF" function can be turned on via editing the yappu *.cfg file to finish with "CRSF:2" which was shown to me by another user online.)

### Version

2.10.4

### Transmitter

RadioMaster TX12 / TX12MK2

### Operating System (OS)

Windows

### Anything else?

This is my first time submitting a Github issue - apologies if I haven't given correct info or formatting, etc.  Please let me know if any more info is needed.  Thank you.

philmoz, 2024-09-30:
I don't think this is an EdgeTX issue.
The yaapu script is using hard coded number to detect events (e.g. event == 128).
Prior to 2.10 the key event numbers were not standardised and varied from radio to radio. This was fixed in 2.10 and all radios now use the same key numbers; but it meant that some values changed.
The script needs to be updated.

You can also use the yaapu configuration script in the TOOLS menu.

d-papworth, 2024-10-01:
Thanks @philmoz . I'll try tagging @yaapu here to see if we can get any clarification.

> You can also use the yaapu configuration script in the TOOLS menu.

I believe I know what script you're referring to - I have it on a TX16S with yaapu that I haven't updated for probably at least a year, I have that configuration script as a separate item in the TOOLS menu on the TX16S - however, I don't get that script appearing after installing yaapu on the TX12 for some reason.

yaapu, 2024-10-02:
Hi, there was no such script before,I added it recentely to work around the menu key limits on some radios, it's "Yaapu Config" in the TOOLS folder, please download latest version for your radio and try this "new" solution

d-papworth, 2024-10-04:
@yaapu  Thanks for your time/reply on this.  I just installed the latest version of the script on a Radiomater TX12 MK2 on ETX v2.10.4.  I still am not getting the menu keys to bring up the script as others have mentioned, but I now have the "Yaapu Config" working great in the Tools menu, and in my opinion that's a great solution. Thanks so much - your work on the script is so valuable to many pilots in the community!

closed 2024-10-04 by d-papworth