EdgeTX/edgetx #5576: Kingtech (and Jetcat) Telemetry Sensors from Spektrum Rx

crwardlaw, 2024-09-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When using a Spektrum Rx, Kingtech turbine telemetry kiddy-backs on the Spektrum built-in Jetcat Telemetry structs. However, EdgeTx does not import the STATUS field which is based upon enums. Kingtech and Jetcat use different enums. My Radio master TX16s Max with Spectrum Rx detects the majority of the Jetcat/Kingtech Telemetry data but not status. I have linked useful documents in the additional context section. N.B.: the status enum for Kingtech seems to use two enums depending on whether the first enum is a specific value. 

The status enums may also be applicable to other Rx protocols, this may not only benefit Spektrum users. 

### Describe the solution you'd like

Preferably option 2 below so that all telemetry and sensor information is held on the one page on the UI.

### Describe alternatives you've considered

1. Just populate the status with its numeric value, and have a specialized "value" widget to display the equivalent text for that numeric status.. like a "valueKTEnum" widget or "valueJCEnum" widget.
2. Create separate sensors for the KT and JC status, populating them with the proper translation. So you pick wich one to use on the screen.
3. Use a specialize widget to display a telemetry screen with all the tubine info together (similar to the spektrum telemetry screen).. the translation of the numeric status is done there.. so one wiget for KT, and one widget for JC.

### Additional context

Spektrum's X-Bus file showing Jetcat enum and Jetcat struct: https://github.com/SpektrumRC/SpektrumDocumentation/blob/2a8a5f9d6f85eff0d2458b9ac2b05e9dcf7ddcab/Telemetry/spektrumTelemetrySensors.h#L739

EdgeTx Spektrum telemetry file showing a TODO for adding status telemetry: https://github.com/EdgeTX/edgetx/blob/149e1a2df00f680404691dadbb59de62abecc3bf/radio/src/telemetry/spektrum.cpp#L193

Document showing the differences between enums for Jetcat and Kingtech - page 41 section 2.9.4: https://drive.google.com/file/d/1TfaMCA3zpDcdGQMnRq5MnjJexTBp-VZD/view

frankiearzu, 2024-09-30:
We could aproach it in three ways:

1. Just populate the status with its numeric value, and have a specialized "value" widget to display the equivalent text for that numeric status.. like a "valueKTEnum" widget or "valueJCEnum" widget.
2. create separate sensors for the KT and JC status, populating them with the proper translation. So you pick wich one to use on the screen.
3. Use a specialize widget to display a telemetry screen with all the tubine info together (similar to the spektrum telemetry screen).. the translation of the numeric status is done there.. so one wiget for KT, and one widget for JC.

Options 1,2 will allow you to log the value into telemetry logs...
Option 3 don't log, but probably looks nicer, and can be done without touching the firmware code.

I see in the Google Drive document that there is already an OpenTX telemetry page.. will research that.. could be a quick way to start.

frankiearzu, 2024-09-30:
Wirdget/Script like this could be the fastest:
[image]

Seems like the link to the code is no longer working.. will try to find it.

crwardlaw, 2024-10-01:
Thank you so much!

frankiearzu, 2024-10-07:
Here is the zip file for the Widget ported to EdgeTX.   
[VSpeak_ECU_EdgeTX.zip](https://github.com/user-attachments/files/17285430/VSpeak_ECU_EdgeTX.zip)
Copy the VSpeak_ECU folder into your WIDGET folder:
[image]

Then on the telemety page, create a new page (full screen, only 1 big cell) and add the VSpeak widget.. and select the ECU (5=JetCat, 6=KingTech).
There is a read_me.pdf with instructions for OpenTX, but is very similar on EdgeTX.

This is how it looks:
[image]

The data comes from telemetry values:
Tmp2 is the Status, not a temperature (need new firmware.. will attach it soon).
[image]

crwardlaw, 2024-10-08:
That's amazing - thank you for putting this test together so quick! I'll be back with the jet this weekend so will try it out and bring feedback on Monday at the latest. 

Did you have to take in consideration the two status enums that kingtech uses where you refer to one when the other is a specific value? I believe that's how it's meant to work.

frankiearzu, 2024-10-09:
I only ported the LUA that they provided, and change the firmware to send the status on sensor Tmp2… so is the same logic that the did provide on their original lua.

crwardlaw, 2024-10-09:
Their LUA script is for use with their proprietry VSpeak telemetry unit which I think may pass the status data differently from the official kingtech and jetcat units, I'll give it a go this weekend and see how it goes but I think the following will also need taken into consideration:

1. The VSpeak unit seems to be treated as a signed int enum (https://github.com/thomasekdahlN/jeti/blob/fbf69aa66e8651a4ddddf9da2f5e5ec90dd9ad4a/ecu/converter/vspeak/kingtech/config.jsn#L15)
2. Whereas the kingtech and jetcat units use 2 unsigned int enums, **status** and **offCondition** which are at lines 611 and 617 in the spektrum telemetry file linked below. When, line 611, **status** == 0x00 then the _reported_ status should instead be taken from the **offCondition** enum, line 617. https://github.com/SpektrumRC/SpektrumDocumentation/blob/2a8a5f9d6f85eff0d2458b9ac2b05e9dcf7ddcab/Telemetry/spektrumTelemetrySensors.h#L611
3. status enum can be found at https://github.com/SpektrumRC/SpektrumDocumentation/blob/2a8a5f9d6f85eff0d2458b9ac2b05e9dcf7ddcab/Telemetry/spektrumTelemetrySensors.h#L621
4. offCondition enum, applicable when status == 0x00, can be found at the link below. It also states when to use this enum further along the line in a comment. https://github.com/SpektrumRC/SpektrumDocumentation/blob/2a8a5f9d6f85eff0d2458b9ac2b05e9dcf7ddcab/Telemetry/spektrumTelemetrySensors.h#L739

In summary, depending on which variable in the struct in the spektrum .h below, is now coming in as Tmp2, I may get one of the status infos coming through https://github.com/SpektrumRC/SpektrumDocumentation/blob/2a8a5f9d6f85eff0d2458b9ac2b05e9dcf7ddcab/Telemetry/spektrumTelemetrySensors.h#L607 

I think EGT will be coming in as Tmp1, therefore, if Tmp2 is coming in as the next variable in the struct, I reckon I will get the offCondition enum coming through to the WIDGET.

frankiearzu, 2024-10-09:
I see, i was thinking that you have the VSpeak Translator.
Im not processing the offStatus code right now, i think what they are doing is passing the OFFstatus as negative.
I can do something similar in the EdgeTX firmware.
Another thing that im not sure is if the value is BCD encoding or not... every other field is documented as BCD, but the statuses don't have any documentation that say BCD encoding.

Lets see what telemetry sensor values you get from your ECU.
What are you using to send the data from the ECU to the RX?? any translator (Like VSpeak)??  just want to make sure it can send the Spektrum JetCap ECU messages,

If the sesors are not OK,  I have a small LUA script to capture the RAW data comming from Spektrum telemetry data. we can try that after the first attempt.

frankiearzu, 2024-10-09:
This is how the code in the firmware is now:
[image]

The status comes as Tmp2

crwardlaw, 2024-10-10:
The kingtech telemetry unit which I have can be found here https://kingtechturbine.lu/KingTech-T-Modul-5-in-1-Telemetry_1
The kingtech telemetry unit connects to the kingtech ecu and the spektrum Rx. I've been able to see it working as have had it hooked up to my Dad's spektrum which was able to receive the temps etc. and seemed to be receiving the status when set to jetcat, but might not have been using the offCondition as that might only be a kingtech thing - it's difficult to tell. I have been in touch with Kingtech and they were not able to provide documentation, so I have had to piece it all together myself.

I think the recording LUA script would be useful to see both the **status** and **offCondition**
 enums changing.

frankiearzu, 2024-10-12:
Did a bit more work on both the LUA and Firmware.
Is combining the Status and offStatus in the way that the LUA widget needed..  (status positive, offStatus negative).
Moved the status to Tmp4 to avoid conflicts.

Tmp3   Turbine Temperature
Tmp4   Status
A3       controller voltage
A4       pump votage
RPM   RPM

[VSpeak_ECU_and_Firmware.zip](https://github.com/user-attachments/files/17349372/VSpeak_ECU_and_Firmware.zip)

crwardlaw, 2024-10-13:
So I have good news and some not so suprising news - firstly thanks for your efforts so far - I can't believe how well this has went!

1. To the best of my knowledge the telemetry comms are working and numbers are matching up where I expect for at least the Rx and Turbine voltages and Temperatures (room temperature), I haven't been able to check the other scalars yet (rpm etc.).
2. Status and offCondition seem to be being detected correctly.
3. However, the enum which the VSpeak widget uses is not a 1:1 mapping to Kingtech nor Jetcat's enums when passing through a Spektrum Rx. Therefore, we're needing to add another enum for Kingtech [and in theory Jetcat if doing the full job] somehow. I'm happy to build that enum as a trial within the Lua Script to make sure I am confident in my thoughts; but, it'll be next weekend before I can again test it. This does lead us to a bit of a hurdle: this would be adding functionality to the borrowed lua script that wasn't for the technology that the original lua script exists for. If I do create the enum for the Lua Script, it will also need to be written to be compatible with the Lua Script's expectations of negative values, rather than the true Kingtech/Spektrum status enums. Like I say, I am happy to change the Lua script to make sure I see the info coming through as I expect, but this now seems like it's moving towards a kludge between the firmware importing the status data to suit a widget's enum which isn't correct for the manufacturer's enums. Do you have a recommended path forward?
4. A tad separate, regarding the firmware you gave me for trialing this: is it expected to be stable or am I best going back to the released firmware I was using for flying in the meantime? I never saw any issues with it, just thought it's better to be safe before having the risk of losing a plane.

Thanks again, Chris

frankiearzu, 2024-10-14:
Hi Chris,
Good that the firmware seems OK, thats the most dificult part.. the LUA is easier to change later.
The firmware was build using the 2.10 branch, that is a pre-release of  2.10.5.  So once we are OK with the firmware, I can create a Merge Request to be added offcially to the next version of EdgeTX.. maybe 2.10.6. I can build a 2.10.5 + this change only later once 2.10.5 is released.

I modified to LUA to add ECU type 26, that will be Kingtech with the Kingtech telemetry module. The name shows as "Kingtech KT Tel".
You can edit the messages on the "msg_table_Kingtech_2" that currently is a copy of the Kingtech ones.
I want to leave the script backward compatible with the VSpeak.

[image]
[image]

New Widget, you can edit main.lua with your favorite editor:
[VSpeak_ECU.zip](https://github.com/user-attachments/files/17359005/VSpeak_ECU.zip)

crwardlaw, 2024-10-20:
I tried the widget and telemetry again over the weekend and I am unfortunately stumped at the order of the enum texts that are required. As there is no clear mapping between the Jetcat status enum : Kingtech status enum : what a spektrum Tx shows for the Jetcat enum : the Kingtech ECU screen status display : vspeak's enum, it is making it difficult to work out what the correct enum should be for the vspeak widget. However, I plan on running the turbine next weekend and that will let me match up more of the status enums between the Kingtech ECU screen and the vspeak widget which will hopefully guide me to get more of them right. I'm also going to get in touch with Kingtech again to see if they can assist by providing the enum that the Kingtech screen uses as that is really what I need. 

Anything else you are looking for at the moment?

frankiearzu, 2024-10-22:
Not much that i can do on my side..   the only thing that i was looking is at the Jetti Script.   Looking at the status messages on Digitech and vspeak converters for Kingtech. Digitech all positive, and vspeak has negative ones and positive ones.
Not sure what the Kingtech converter sends as status on the Spektrum Telemetry Frame.

On the spektrum Radio, it it was made for Jetcat, it will be showing you probably the wrong status message.

Looked at a few of the configurations of the Jetti Widget:
[image]
https://github.com/thomasekdahlN/jeti/tree/fbf69aa66e8651a4ddddf9da2f5e5ec90dd9ad4a/ecu/converter

crwardlaw, 2024-10-26:
Hi!

So i have managed to confirm a few more of the status enum values this weekend by running the jet. Unfortunately I think a few may be missing due to the jetcat enum which I believe it is based on - such as no "cooling" value. Thought I was able to see a number of values with similar descriptions which has given me confidence that I am on the right track.

With this further testing I have discovered that I am unable to check whether the offCondition is working, and in fact, it may not be.

1. May you please set up the offCondition to come through on its own telemetry as well in the settings, so I can see it changing on its own as well as it being used in the current implementation, so I can see if it's changing as expected. Even another one which is just the status enum to be able to see them both separate and combined.
2. Am I able to use the DSM telemetry recorder to record this telemetry? I ran it and wasn't certain it was recording all the values, looking at the code too.
3. I have been in touch with Kingtech and Spektrum for further assistance as Kingtech believe it should be fully functional with the most modern Spektrum Txs, which would suggest there's functionality that isn't on spektrum's github like the rest of their documentation.

Thanks,

Chris

frankiearzu, 2024-11-08:
Sorry for the delay, did not see your response.. Populated the Status on 'TMP4' and offStatus in 'TMP5'.

[tx16_2.10.5_jetcat2.bin.zip](https://github.com/user-attachments/files/17671810/tx16_2.10.5_jetcat2.bin.zip)

frankiearzu, 2024-12-28:
Chris, any news on this? Have you been able to edit the enums to match?

crwardlaw, 2024-12-28:
Hi I was thinking about this this week. We've had a change of flying sites and so I'm not going to get the jet back out again until April at the earliest. What we have managed has been great and a big help so I think it's definitely worth closing out in some form. I'll come back with another response in the next couple days with what I've got.

frankiearzu, 2024-12-28:
Ok, I will create a Merge Request for the firmware with this change and other improvements/cleanup that I have for Spektrum telemetry.  Everything else we can handle it on the LUA script later.

crwardlaw, 2025-01-19:
### Summary

1. Through multiple start up and shutdowns of the kingtech turbine I was able to get basic feedback from the turbine telemetry correctly displayed on the transmitter.
2. I was not able to get the full range of status values which the turbine display shows to the transmitter for a number of reasons.

### Process

1. I found that for the most part the enum being used in the existing luas did not match what the turbine was using. This is expected as the script was written to work with a third-party telemetry unit.
2. To confirm the specification of the enum I got in touch with kingtech and they did not provide the enum they use. When I pointed out that I believe they were not passing through all the available status information and that the enum could be expanded, they did offer for me to send back my kingtech telemetry unit to have an updated firmware put on it. I do not know if they have since formally released this updated firmware. I did not follow up on this conversation as I was hesitant to send the unit back. I may follow up with this in the future.
3. To confirm the enum kingtech were using I had to carry out multiple start ups and correlate the enum values the transitter received with what the kingtech turbine display showed. Then rewrite the enum in the script through this matching.
4. This new enum is specified within the attached file. I have added a comment to values in the enum, stating any current feedback I have on it.
5. Confirming the error cases throughout the enum is difficult as I cannot induce the errors to see what values they are recorded as.
6. Using the updated firmware, Tmp5 remained at 0 throughout an entire run of startup -> succesful run -> succesful shutdown. Therefore, I have no evidence of the negative enum values coming through correctly.

### Next Steps

1. It seems safest to comment out the unconfirmed values, incase anyone was to use this and get incorrect feedback.
2. I do not have evidence that the negative enum values are being displayed correctly, so I cannot be sure that logic is working correctly. Although previously I did think it was, but now cannot find the specific evidence of that.
3. Update the telemetry widget to display both the Rx and the Turbine battery voltages. It currently only shows 1 (it has slipped my mind, but I'm sure it is the turbine battery that it currently displays).

### Attachments

Widget Script with Updated enum and commented assignments in enum: [250119_Telemetry_widget_CW.zip](https://github.com/user-attachments/files/18470524/250119_Telemetry_widget_CW.zip)

aeroxfrank, 2025-05-29:
Yeah thats right but you can use serial port on it and it has a dedicated serial port where you can use hot telemtry on!