EdgeTX/edgetx #5577: HelloRadioSky V16 - external Modul bay

Futaba-Guy, 2024-10-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I trieb the 2.11- nightly at my HR V16pro. After flashing the external Modul HappyModel EX24pro doesn‘t get power. So the lua script eLRS didn‘t work. I open the Modul. Inside is an USB-Port. I powered the modul while sitting in the bay. It worked. So I flashed back to delivery state (2.10.1-selfbuild) and all worked fine again.

### Expected Behavior

Thai the modul gets (enough) power

### Steps To Reproduce

Everytime I put the modul in or fired up the transmiter I doesn‘t work with 2.11-nightly

### Version

Nightly (Please give date/commit below)

### Transmitter

HelloRadioSky V16

### Operating System (OS)

Windows

### OS Version

11

pfeerick, 2024-12-05:
Can you try again with a recently nightly. I am unable to reproduce this with latest nightly and RM Ranger ELRS module - module receives power and can be communicated with via ELRS Lua.

Futaba-Guy, 2024-12-06:
Hello Peter,I can’t find any issues with the nightly 366df608 anymore (I don’t know wich nightly this is). With today’s 3d1e7a9 is also all working.Kind regards Thomas

pfeerick, 2024-12-07:
Thanks for checking, and for the update. I'll close the issue then as it looks to have been resolved.

closed 2024-12-07 by pfeerick