EdgeTX/edgetx #5581: [MT12] Inconsistency in trim position and name

chofchop, 2024-10-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In the current MT12 firmware, trim 1 is assigned to trim RH on the screen, and trim 2 is assigned to trim LH on the screen.
In this case, the physical trim position and the trim position on the screen are reversed left and right, so it is very unnatural and cannot be operated intuitively.
Also, on the UI, trim 1 is named "ST" and trim 2 is named "TH", so if the display positions of trim 1 and trim 2 are reversed horizontally, the trim names will be inconsistent.

### Expected Behavior

Display trim 1 at trim LH and trim 2 at trim RH on the screen.

Name trim 1 as "T1" and trim 2 as "T2" (do not use "ST" or "TH").

### Steps To Reproduce

All firmware after the official release of EdgeTX v2.10 seems to have the same specifications.

### Version

2.10.4

### Transmitter

RadioMaster MT12

### Anything else?

It is obviously unnatural that the trim position is horizontally reversed, so it needs to be corrected.

Unlike stick radios, there is no clear correlation between each input axis and each trim switch on the MT12. 
There are markings “T1”, “T2”, “T3”... on the MT12 body. Calling them "ST", "TH", "T3"... on the UI causes confusion.
In fact, the MT12's trim switch is often used for purposes other than trim, and throttle trim cannot be used. (I'll explain later). I don't think it's a good idea to use "ST" or "TH" in trim names.

3djc, 2024-10-02:
Done on purpose

chofchop, 2024-10-02:
If you don't mind, could you please tell me the purpose?

chofchop, 2024-10-02:
Thank you for answering. You're talking about Trim's name.

What do you think about the physical and on-screen positions of T1 and T2 being flipped horizontally?
Was this also done on purpose?

3djc, 2024-10-02:
Yes, our initial implementation was different, but we got feedback that it was unnatural for seasoned surface drivers

chofchop, 2024-10-02:
Thank you, I understand. But I'm very disappointed.

chofchop, 2024-10-02:
This is off-topic, but regarding the matter I said "I'll explain later" above.

For RC cars, it may be necessary to move the neutral position of the throttle (drag brakes, etc.). This requires Δ mode (Center Only) trim. In EdgeTX, Δ mode can only be used for output line sub-trim.
Therefore, throttle trim cannot be used directly and must be used via the GV assigned to sub-trim.

inventor7777, 2024-10-25:
I also wish the first two trims could be renamed to just T1 and T2, but its pretty minor.