EdgeTX/edgetx #5597: Helloradio v16

Wesselink1966, 2024-10-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

Channel 5 and 6 become un use able after voice recognition is enabled  (lua on).
When (lua) off all functions ch5 and 6 normal

### Expected Behavior

 I expect full use off channels with voice recognition enabled

### Steps To Reproduce

Set lua voice  on in settings . Move slider in menu. Ch5 and 6 unusable.
Set lua voice off all is ok.

### Version

2.10.4

### Transmitter

HelloRadioSky V16

richardclli, 2024-12-04:
There are 2 versions of firmware for V16.  Their stock firmware has quite a lot of hard-coded features to demonstrate the capabilities of their new V16 radio.  However, these new features need time to become part of EdgeTX code base.  Current 2.11 nighties has V16 support with limited functionalities.  V16 new features will be added one by one in the future development with join effort of EdgeTX team and HelloRadioSky team.

Edwin-Camargo, 2024-12-04:
My HelloRadio arrived this past Saturday, but my joy was short-lived: it responded normally when I said “HelloRadio”, but shortly afterwards it stopped responding. Maybe I changed some settings by mistake, but I think I did what I was supposed to do.
 
The radio is an ELRS version, and is configured as follows:
a) Speech recognition Management: Voice = LUA and Power Switch = ON (enabled);
b) CH56 Release: I tried both ways: on and off
c) I have already updated the radio’s firmware with the version available on your website (V16-firmware-2.10.1-092401-EM.bin) and it didn’t work.
d) I have already deleted a model I created (a model aircraft I created for testing) and it didn’t work either.

I don't know what else to do. Can recording another microSD card solve the problem?

Could someone help?

pfeerick, 2024-12-05:
So when you say the trigger word "HellowRadio" the radio does not respond "Hello"?

philmoz, 2024-12-05:
I thought the voice recognition was only in the custom firmware delivered on the radio - it hasn't been added to our code base yet.

pfeerick, 2024-12-05:
That goes without saying... the options are only available with the custom firmware he is running per the screenshots ;)

philmoz, 2024-12-05:
[image]

pfeerick, 2024-12-05:
Which is the custom firmware that is in the SD card bundle from the https://helloradiosky.com download page. We (EdgeTX) do not (and never have) made firmware available with those filenames ;)

Edwin-Camargo, 2024-12-05:
> So when you say the trigger word "HelloRadio" the radio does not respond "Hello"?

Exactly, it stopped responding :/

Edwin-Camargo, 2024-12-05:
> I thought the voice recognition was only in the custom firmware delivered on the radio - it hasn't been added to our code base yet.

I don't understand much about this. It's my first radio of this type, but I guess you're right.

Edwin-Camargo, 2024-12-05:
> Which is the custom firmware which Richard referred to above that is in the SD card bundle from the https://helloradiosky.com download page. We (EdgeTX) do not (and never have) made firmware available with those filenames ;) Neither is it supported in an "official" 2.10 release - limited support is only present in `main` (i.e. for v2.11).

This is the version I downloaded from the manufacturer's website --> **V16-firmware-2.10.1-092401-EN**

Edwin-Camargo, 2024-12-05:
I'm thinking about burning a new microSD card with the manufacturer's files and seeing if it solves the problem, but I'm VERY afraid it might be a problem with the voice recognition chip...

pfeerick, 2024-12-05:
Ok, so with the "V16-firmware-2.10.1-092401-EN" build, my V16 is working fine. I only flashed the firmware, I've not updated the rest of the SD card pack yet. If you check your SD card's firmware folder, you may find you still have the older firmware your radio shipped with, and can try flashing that again. i.e. hold the horizontal trims (T4, T1) inwards, and power on the radio, choose write firmware, and pick another firmware file (preferably ending with EN if you want the "En"glish version). 

Based on your earlier screenshot, you're not running the "V16-firmware-2.10.1-092401-EN" firmware at the moment - but a slightly older build from August (2024-08-25), rather than 2024-09-24 (i.e. month and year in the filename). If voice recognition works again, it may be a glitch with that firmware build that you are running. Also please note you need to wait 5-6 seconds after the radio has booted up / voice recog has been powered on, for it to become ready for use - it takes a few seconds to boot itself up.

Edwin-Camargo, 2024-12-06:
> Ok, so with the "V16-firmware-2.10.1-092401-EN" build, my V16 is working fine. I only flashed the firmware, I've not updated the rest of the SD card pack yet. If you check your SD card's firmware [...]

Okay, thanks. I'll check that out.

Edwin-Camargo, 2024-12-06:
Eureka!!!! SOLVED IT!!!!
I was also in contact with Engineer Ken (from HelloRadio itself). He gave me the step-by-step guide to solve the problem.
To summarize:
a) I replaced the microSD card (at my own expense, it wasn't part of the suggestion);
b) I saved all the factory files on it again;
c) I updated the radio's firmware to the factory one, specifically the one from 09/24/2024 (which appears in the file as 092401);
d) I updated the voice recognition chip's firmware (a bit of a tedious process, but...).

I thank Ken and the other friends who responded here for their help!

SOLVED!!!

I just didn't understand why the problem occurred after only 20 minutes of using this radio, anyway.

pfeerick, 2024-12-07:
Fantastic to hear! Glad you got it working. I'll delete all the comments in this issue from https://github.com/EdgeTX/edgetx/issues/5597#issuecomment-2518574426 onwards in a day or so (and move your last one to a discussion thread so others can find that info if they face the same problem) as this off topic to the issue started by Wesselink1966 (which was specifically that CH5/CH6 were unusable once voice recognition was enabled).

Wesselink1966, 2024-12-07:
5 and 6 and voice recognition works only with original hello radio firmware. Problem solved. Tnx

cptX, 2025-09-28:
So channels 5 and 6 now work when voice recognition is activated? Can anybody tell me if after these months there is any integration of this radio with the official EdgeTx Firmwares?