EdgeTX/edgetx #5598: Is there a wrong Lua compiler in Companions Simulator ?

arrenbrecht, 2024-10-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Telemetry Lua scripts with float variables are aborted in the Companion Simulator with the following error message
mailformed number near '.1'

I suspect that an LUA compiler is included that can only process integer variables.

LUA scripts on the transmitters run flawlessly.

### Expected Behavior

Lua scripts with float variables should also be able to be started in the Companion Simulator.

### Steps To Reproduce

start any LUA script with float variables

### Version

2.10.4

### Transmitter

RadioMaster Pocket, RadioMaster TX16S / TX16SMK2, RadioMaster Zorro

### Operating System (OS)

Linux

### OS Version

Linux Mint

pagrey, 2024-10-07:
It might help if you post the code that is giving you problems. All of my scripts run fine on 2.10.4 in the simulator.

wimalopaan, 2024-10-08:
You need to set the env variable `LANG=C`

arrenbrecht, 2024-10-08:
Where and how do I have to set the variable?

wimalopaan, 2024-10-08:
My knowledge is limited to linux:
in your `shell`:
`$ export LANG=C`
Afterwards start your simu/cpn from this shell.

arrenbrecht, 2024-10-08:
Thank you, it works

closed 2024-10-08 by arrenbrecht