EdgeTX/edgetx #5603: Change LED Color for 6-Pos on RM Boxer

katomic, 2024-10-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Since the Power button's LED can be Red, Green or Blue, I was wondering if the RM Boxer's hardware has a mulitcolor LED under the 6Pos buttons? 

### Describe the solution you'd like

If the 6Pos LEDs can indeed be individually addressed, it could be used as an alternate information display available on the Special Functions page via the UI ( similiar to Play Track, or  option for example).
Since the 6 LEDS flash seperately during boot, this seems likely to be possible.
Building on that, allowing each LED to be switched on/off, blink, change color could be implemented.

### Describe alternatives you've considered

1. Turning individual LEDs in the 6 POS on/off
2. Changing color of individual LEDs in the 6Pos

pfeerick, 2024-10-10:
Not what you were hoping to hear, hut neither of these are possible with the RM Boxer. None of the 6POS LEDs are controllable from EdgeTX. Only Jumper handsets with customisable switches have controllable leds there at the present time.

closed 2024-10-10 by pfeerick

katomic, 2024-10-11:
> Not what you were hoping to hear, hut neither of these are possible with the RM Boxer. None of the 6POS LEDs are controllable from EdgeTX. Only Jumper handsets with customisable switches have controllable leds there at the present time. And the Boxer does not have RGB LEDs on 6POS.

I appreciate the feedback. Is this limitation due to firmware on a circuit other than the one running the edgetx fw? Being a dev myself, thought that the strobing of the 6 LEDs on power-on means that they are definitely controllable, perhaps just not from the main board?

pfeerick, 2024-10-11:
> due to firmware on a circuit other than the one running the edgetx fw

This is exactly the case. The 6 button 6POS "switch" on pretty much all of the radios is done via a separate MCU, that reads the input from the buttons, controls the LEDS, and then simulates an analog output which the main handset MCU can then read. You can see more in risto's [reverse engineering of the chip used on the RM TX16S](https://www.rcgroups.com/forums/showthread.php?3855505-Blog-13-RadioMaster-TX16S-6-Position-Switch-firmware), which is most likely the exact same one used on the RM Boxer. 

You might also be interested in some of his other posts about this handset, for context:
- [#13 RadioMaster TX16S 6-position switch firmware (written from scratch)](https://www.rcgroups.com/forums/showthread.php?3861617-Blog-15-RadioMaster-TX16S-breakout-board)
- [#17: RadioMaster TX16S schematic diagram](https://www.rcgroups.com/forums/showthread.php?3869543-Blog-17-RadioMaster-TX16S-schematic-diagram)