EdgeTX/edgetx #5604: Timer Countdown not working

ecs05, 2024-10-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If the timer is set to Countdown ,Start 20 sec., (Voice to 5 
sec.)the message Countdown = reached appears as soon as the timer is started. Timer display 0 seconds, the timer field flashes red and the time continues to run in the plus direction, start value 0. This means that the timer never runs down from e.g. 20 to 0 seconds and there is a voice warning at e.g. 5 seconds in any setting.

### Expected Behavior

The Time Countdown start wit 20 Sec. und go down to 0 Sec. e.g. minus if the timer is not stopped

### Steps To Reproduce

Timer 1
Mode : on
Switch : SG-
Start   : 20 Sec.
Direction : Show Remain
Countdown : Voice
Persistent : manual Reset

### Version

2.10.4

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### Anything else?

[image]

philmoz, 2024-10-11:
Not reproducible in simulator or on my TX16S.

Please zip up your MODELS, RADIO, SCRIPTS and WIDGETS folders and upload them here.

philmoz, 2024-10-11:
It works fine for me in the simulator:
- Set SG to UP
- Press ENTER
- Select 'Reset Telemetry'
- Select 'Reset timer1' (timer is set to 20 seconds)
- Set SG to middle

Timer starts counting down from 20.

You have a special function that resets timer 1 to 0 when SG is DOWN - that is probably what is clearing the timer,
[image]

ecs05, 2024-10-11:
Ohh- Looks a lot like my mistake.
Sorry and thanks for the quick help

closed 2024-10-11 by 3djc