EdgeTX/edgetx #5607: Issue with EM handling in main

3djc, 2024-10-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After EM happens, SD is mounted, sound is available, I guess LUA possibly too

### Expected Behavior

In EM reboot mode, SD must stay unmounted, to prevent entering a EM loop sequence

### Steps To Reproduce

Trigger EM on a debug build

### Version

Nightly (Please give date/commit below)

### Transmitter

Other (Please specify below)

### Anything else?

Found this testing future hardware, but the issue is not hardware related. I have seen it in BW, I have not checked colorlcd (nor have I tested 2.10). I will not be able to work on this for at least 5 or 6 weeks

philmoz, 2024-10-12:
Are you building without RTC_BACKUP_RAM set?

Looking at the code the behaviour you describe would only happen if RTC_BACKUP_RAM is not defined.

3djc, 2024-10-12:
Is it not defaulted for all radio ? I'm at least not excluding it

philmoz, 2024-10-12:
It should be; but looking at the code this is the only reason I can see why the SD would be mounted.

3djc, 2024-10-15:
Tested Boxer, same issue, so my guess it affect all BW radios

philmoz, 2024-10-15:
It would appear the RTC_BACKUP_RAM is not defined for B&W radios. It look the same in OpenTX.

Since there is no backup what else can EM do; but load the model settings from the SD card?

3djc, 2024-10-15:
Yeah but OTX had models in eeprom. The issue is you can run into boot loops if the source of the EM is sd card related. Need to assess what we can do, maybe limit sd access to bare minimum

3djc, 2024-10-16:
I have successfully run RTC_BACKUP_RAM based EM on a RM Boxer, will do a PR for BW F4 when I'm physically capable

PR #5686 (2024-12-05): feat: use RTC backup for EM on BW F4 1Mb flash radio
On F4 based BW radios with 1MB flash, use backup register for Emergency Mode handling (like it is done on colorlcd) instead of relying on SD card. As a side benefit it shortens EM time reboot by nearly a second by not mounting SD card.

This fixes #5607
closed 2024-12-05 by pfeerick