EdgeTX/edgetx #5608: lcd.sizeText(XXLSIZE) changed it's vertical position on LVGL compared to LCD version

offer-shmuely, 2024-10-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

in LVGL the 32px font (XXLSIZE) is placed asymmetrically on the vertical line,

### Expected Behavior

in LVGL the 32px font (XXLSIZE) should be placed symmetrically on the vertical line
the same as non-lvgl version

### Steps To Reproduce

```
local function update(wgt, options)
    wgt.options = options
    lytZone2 = {}
    local text = "FONT XXLSIZE"
    local ts_w1, ts_h1 = lcd.sizeText(text, XXLSIZE)
    lytZone2[#lytZone2+1] = {type="label", text=text, x=10, y=100, font=XXLSIZE, color=WHITE}
    lytZone2[#lytZone2+1] = {type="rectangle", x=10, y=100, w=ts_w1, h=ts_h1, color=BLUE, filled=false}
    lvgl.build(lytZone2)
end

```

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

win11

### Anything else?

[image]

offer-shmuely, 2024-10-12:
[image]

[image]

philmoz, 2024-10-31:
This is because the legacy 'lcd' text drawing function adds an offset when drawing the XL and XXL font characters.
The lvgl text rendering does not add this magic offset.

The comments indicate this was done for compatibility with 2.3.x.

The magic offset is also added to the 'getTextSize()' function for the XL and XXL fonts.

Currently text drawn with XXL looks like this (note the waste space at the top):
[image]

It should look like this:
[image]

IMO this magic offset should be removed, although this may change the way some existing scripts look.

offer-shmuely, 2024-10-31:
since anyone using lvgl is changing code anyway, I see no issue of backward compatibility here, 
So it is better make the textSize as accurate as possible.

PR #5671 (2024-11-24): chore!(color): remove font offsets for Lua 'lcd' text drawing
Make text drawing consistent between firmware, Lua 'lcd' drawing and Lua 'lvgl' drawing.

Fixes #5608 

Note: this may affect the visual display for some widgets.
closed 2024-11-24 by pfeerick