EdgeTX/edgetx #5609: lcd.sizeText() is not accurate

offer-shmuely, 2024-10-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

### background:
I am trying to write a widget (re-write the BattAnalog), in a way that it will be fully adaptive to different widget zone
This way we could use it also in different TX resolutions
placing the text while trying to maximize the usage of the widget real estate is a hard task, 
I am using the lcd.sizeText(), but it must be accurate to achieve a good results.

### the issue:
lcd.sizeText() is too conservative, it take a lot of spares

[image]

### Expected Behavior

lcd.sizeText() should return the exact size of the text given

[image]

### Steps To Reproduce

sample widgets: 
[WIDGETS.ZIP](https://github.com/user-attachments/files/17349482/WIDGETS.ZIP)

### Version

2.10.4

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

win11

offer-shmuely, 2024-10-12:
that is exactly the bug :-)
the function does request my string, and it does calculate the width based on my string
I do expect it to calculate the height as well based on my string

3djc, 2024-10-12:
First, not a bug since this is per design: it allows to concatenate text in efficient way, by giving a 'next' point given your current exact text, and a height that will allow this, or text to come to fit

offer-shmuely, 2024-10-12:
Ok, 
So can I get another function, or an additional flag
In such a way the function will calculate the size based on the string I send it(!), 
and not based on theoretical string that will never come?

I like to build a dynamic widget, that will not assume any predefined size.
and we do not have css tools, 
So I like to have that can help me achieve my goal.

philmoz, 2024-10-12:
Unlikely - text is fixed height in LVGL (which is used for both cases). There is no facility provided to get the pixel height of an individual character.

offer-shmuely, 2024-10-12:
tough one :-(
Lets try from other directions,
I notice that all the A-Z a-z are the same height as a typical "A" height.
Why the special characters are different height?
Is changing the font, only on the special characters, is an option?

philmoz, 2024-10-31:
The character heights are determined by the font source used to create the lvgl fonts.
You also need to remember that the fonts contain characters for non-english translations which are taller.

[image]

offer-shmuely, 2024-10-31:
I like the getTextSize to calculate the size based on the given text, and not on a text that maybe.
It should not be static on the height, the same it is not static for width

philmoz, 2025-08-01:
As I've already explained, the height of the font is a fixed value which is already returned correctly by the lcd.getTextSize function.

closed 2025-08-02 by offer-shmuely