EdgeTX/edgetx #5610: edgeTX 2.1O /+ does not work on my radio

leptitblond404, 2024-10-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

when i turn on my radio it doesn't pick up the sd card. I've tried with several sd cards with the correct file versions.
To make it work normally again, I just have to switch back to 2.9 or lower, but it doesn't pick up the rx anymore.

### Expected Behavior

that everything is running smoothly

### Steps To Reproduce

set the radio to edgeTX 2.10.0 / 2.10.4 and switch it on

### Version

2.10.4

### Transmitter

iFlight Commando8

### Operating System (OS)

Windows

pfeerick, 2024-10-12:
As per the release notes, 2.10.3 was the first version of 2.10.x that was
safe to use with the Commando8. Prior versions of 2.10.x had a power-on
issue rendering the handset unusable as it would just shutdown, hence the
warning to not update.

What exactly do you mean by "it does not pick up the SD card" ??

pfeerick, 2024-10-13:
I cannot reproduce this with iFlight Commando8 running v2.10.4 - SD card is working fine, and am able to view it via PC after entering USB connect mode (unique to the Commando8 - hold down the power button after plugging the USB cable into the PC to change it from charge mode to PC connect mode).

leptitblond404, 2024-10-15:
I saw that versions 2.10 to 2.10.3 were not compatible with my radio.
what i mean by “it doesn't detect the sd card” is that if i turn on my radio in version 2.10 + it displays “no sd”.
and to switch to dfu, just connect to the pc, press the menu button downwards and press the power button.
sorry for the late reply.

pfeerick, 2024-10-15:
Thanks for the update, so this is specially 2.10.3 onwards, not 2.10+

Most likely, if your SD card worked with 2.9, but now doesn't work, you'll need to replace the SD card as there have been some changes that make cards that are extremely slow to respond to initialisation unusable. i.e. if you have a blank name brand microSD smaller than 32gb handy, FAT (not exist) formatted, try that instead...

leptitblond404, 2024-10-17:
ok thanks I'll give it a try and keep you posted
for info my sd card is a sandisk 8GB, U1.

leptitblond404, 2024-10-29:
everything works fine for me

closed 2024-10-29 by leptitblond404