EdgeTX/edgetx #5614: LemonDSMP: No telemetry, and CH8 not working

pfeerick, 2024-10-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When using the LemonDSMP Module, there is no telemetry from the [LM0086 7CH](https://lemon-rx.com/index.php?route=product/product&path=76&product_id=270) receiver, and it appears that CH8 (and maybe also CH7 as stabilsation should still do something with minimum gain) is not working / locked to  -150. I've encountered this myself, and it has been reported by another on RCG
- https://www.rcgroups.com/forums/showpost.php?p=53053491&postcount=8386
- https://www.rcgroups.com/forums/showpost.php?p=53053971&postcount=8390

### Expected Behavior

For Telemetry to work, and for all available channels to work correctly. 

### Steps To Reproduce

1. Bind LM0086 RX to LemonRX DSMP module with LemonRX DSMP set as the module protocol
2. Attempt telemetry scan - nothing will be found
3. Try to use CH7/8 for stabilisation control/gain - nothing will happen

### Version

2.10.4

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

I encountered this when originally testing 2.10 prior to the first RC, and was able to repo it also in 2.7.1. However, at the same time, I have working telemetry with the LemonRX DSMP module if using a Spektrum AR637T receiver, and if I use the MPM I can get telemetry from the LM0086 receiver. 

Note: this is the first gen LM0086, **not** the second generation LM0086X if that has any bearing on this.

raphaelcoeffic, 2024-12-01:
According to Mike's code on RCG, we were indeed not using the right bit mask (`0x3F` vs `0x47`), which may or may not fix the issue. I don't have anything to test this, so I don't know whether or not this fixes anything.

raphaelcoeffic, 2025-10-06:
@frankiearzu many thx for taking the lead on this one! Let me know if you need anything.

frankiearzu, 2025-10-06:
Some update of this... 
I have been playing with it with the intention of adding forward programming capabilities to it to begin.

During that effort, I noticed that a lot of channel messages are being lost and never passed to the RF sending unit.
Using a logic analyzer to see the data in and out, and the timing.

Probably we haven't noticed it, since all are constantly being refreshed.. but on a Command/Respond message protocol like Forward Programming, any loss of messages is noticeable. 

Talking to Frederick at LemonRX, and looking the Expected message timing of the DSMP module, it shows that the DSMP is expecting data from the TX to be at 22ms cycles.. 

Looking at the EdgeTX code, we are sending on an 11ms cycle... that means that one of the messages in each 22ms cycle will be lost.
Once I changed the refresh cycle/period to 22ms, everything started to work much better. 

A little more graphical timing:  The top was provided by Frederick, and added some annotation of what EdgeTX is doing to help my understanding.

[image]

This s the offending line:
[image]

Once changes to 22ms, it does not lose any messages. 

I thought I could fix much of the forward prog on the EdgeTX side, but needed some changes on the DSMP too.
Also working with Frederik on a V2 of the DSMP firmware to add some status messages to show how the DSMP is working.

So on the EdgeTX side, this is what I have done.. always having backward compatibility to the original DSMP module (lets call it V1), and the new Firmware will be V2.

My changes so far:
1.  Fix the refresh timing from 11ms to 22ms.
2. Send forward programming data from EdgeTX -> DSMP   (the V1 will just ignore the message).
3. Enhancements to the External module screen to show a few extra info.
3.a.   Show Status message  (currently with some extra debugging data for myself).
3.b.  Allow to choose receiver ID, on the V1, is hardcoded to Receiver ID=1, but the V2 Firmare will use what you have.
3.c.  The channel range was weird, it did allow you to change the Starting channel and the End channel was disabled.. Fix so that the starting channel is always 1 (as needed by the DSMP), and only edit the ending channel.
3.d. PENDING:  Will add channel mapping capabilities to convert from the model Ch order (AETR or other) to TAER (as needed by the module).
4.  On the Spektrum Telemetry side (Spektrum.cpp), there are parts  with (if DSMPModule) and if (MultiModule).. to be able to turn off the BIND when telemetry is received.   That Specific module code parts, I move them to the files closer to the specific module (pulses files), and left only on the Spektrum.cpp telemetry file with what is common..  In this way,  the code related to the multimodule is in the multi.cpp and the part related to DSM/DSMP are in the DSM2.cpp file.

NOTE: Some extra debugging data in the status that will be removed.
[image]

[image]

Question for the EdgeTX experts: @raphaelcoeffic @pfeerick 
1.  Currently, the code from DSMP is mixed with the code from the older DSM2 module. I could split the DSMP code to be on its own pulses C/C++ file.  They really don't share much on the sending and receiving data.
2. Make that the code can be included/excluded depending on an #ifdef DSMP, similar to how the MULTIMODULE is.
3. All the code is backward compatible to original DSMP module.   I am planning to do 2 PR.. one with everything compatible with V1 firmware, and a 2nd PR for things that are still backward compatible, but intended for the V2 firmware.   The V1 changes are also compatible to 2.11.x..   @pfeerick Are we planning any patch release for 2.11.x before the offical 3.x??

Haven't tested much the PPM side much, but will spend some time after.   That timing discovery took me a lot of time figure it out!

frankiearzu, 2025-10-07:
Looks like the problem after BIND, which the TX gets into a weird state, is caused by trying  the call to "restartModuleAsync(module, 50);".. I commented out, and it works well after binding.   Before, the only option was the turn OFF/ON the TX.
Reported here:  https://www.rcgroups.com/forums/showpost.php?p=53053971&postcount=8390

pfeerick, 2025-10-07:
Thanks for digging into this! When you have something that you want more testing done with, let me know... I won't be available for the next 1 1/2 weeks though as of pretty much now. ;)

pfeerick, 2025-10-07:
Sorry, didn't see that.

> Currently, the code from DSMP is mixed with the code from the older DSM2 module. I could split the DSMP code to be on its own pulses C/C++ file. They really don't share much on the sending and receiving data.
>
> Make that the code can be included/excluded depending on an #ifdef DSMP, similar to how the MULTIMODULE is.

Yes, If it is unique enough, it makes more sense to have its own file and structure than interleave it with the DSM2 code. And good to be able to disable it if someone is doing a build where space is a premium. 

> All the code is backward compatible to original DSMP module. I am planning to do 2 PR.. one with everything compatible with V1 firmware, and a 2nd PR for things that are still backward compatible, but intended for the V2 firmware. The V1 changes are also compatible to 2.11. Are we planning any patch release for 2.11.x before the offical 3.x??

Yes, there will be more 2.11 patch releases before 3.0. It would be nice to have the DSMP stuff working properly in a "stable" release, so if you want to do a PR against both main and 2.11 branch that will be fine (and appreciated!)