EdgeTX/edgetx #5618: Jumper T16 Pro won't flash 2.10.x

mjdavsge71, 2024-10-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

No matter if I use the buddy site or download the bootloader/firmware I am unable to flash to any 2.10.x version of EdgeTX, the most recent version I have success with is EdgeTX "Providence" v2.9.4.  I was able to flash 2.9.4 using the .bin file from the SD card or using the buddy site.  If there's anything newer that I try, it appears to flash. After the flash, I unplug the cable, but when I go to try to power back on the radio, all I get is flashing white lights at the top. 

### Expected Behavior

 When I run the firmware push from the buddy site, after the flash is complete, I unplug the USB-C cable and start the radio. It powers up and shows me the Edge TX logo and works and operates normally. 

### Steps To Reproduce

To reproduce the problem, I could just try to flash any 2.10 version and it will fail. 

### Version

2.9.4

### Transmitter

Jumper T16

### Operating System (OS)

Windows

### OS Version

11

pfeerick, 2024-10-17:
The Jumper T16 I have access to was running 2.10.3, and I was just able to DFU flash it with 2.10.5 using EdgeTX buddy, with a successful boot afterwards, so not sure what is going on there. 

Has this handset been modified an any way?

If you want to double check, this is the (zipped) firmware I just flashed : [t16-v2.10.5.zip](https://github.com/user-attachments/files/17405514/t16-v2.10.5.zip)

mjdavsge71, 2024-10-18:
Used your zip .bin and flashed it from EdgeTX  Buddy, once the flash completes I remove the USB then hold the power and all I get is the flashing 1 thru 6 lights until it settles on light 1.  I'm re flashing 2.9.4 now.... and it works after the flash.  Seems like its unable to read the new firmware and start with anything over 2.9.4,  Just a T16 Pro no mods.

mjdavsge71, 2024-10-18:
I copied 2.10.5 to the internal SD. After flashing 2.9.4 and restarting the radio, I attempted to flash the 2.10.5 bootloader from the .bin file you provided me with (I did this before using the 2.10.5). Tried one more time flashing to just 2.10.0 using EdgeTX buddy and same result, went back to 2.9.4, scratching my head on why it doesn't like the bootloader on 2.10.x.

pfeerick, 2024-10-18:
Mine is stock config also, so very strange. All I can thing of is one of two possibilities 

1 - new firmware doesn't like something in the RTC backup memory. To resolve that, you'll need to open up the transmitter, so you can pull the RTC battery out. So you don't have so long a wait, you can then try pressing the power button (to drain whatever residual power is left in the capacitors, etc), and then leave the radio a good 15 minutes with no batteries. Reassemble and try booting it with 2.10.x now. 

2 - bad MCU fuse setting. We ran into a problem with the TX16S where it seems either there was a bit flip, or bad setting from the factory, which clashes with 2.10 changes. Check here for the resolution steps for that: https://manual.edgetx.org/installing-and-updating-edgetx/having-trouble-updating#i-updated-edgetx-and-now-my-radio-doesnt-turn-on-or-acts-strangely

mjdavsge71, 2024-10-18:
@pfeerick  Thanks for the assist.  It was option 2 fixed my issue, just unchecked that BFB2 option, it apply, yanked the USB cable and powered it on. Now I see the cool new 2.10.5 porcupine? logo.

pfeerick, 2024-10-18:
Fantastic! And close.... Eddie is a hedgehog 😂

closed 2024-10-18 by pfeerick