EdgeTX/edgetx #5626: RadioMaster TX16s MK I - when setting display off brightness to 0 the display will not return to normal brightness

Astrohog, 2024-10-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Sometimes I leave my Tx on between flight so I decided to set the "System - backlight - off brightness" to zero.  Mode is set to both.  The first time I made the change the radio seemed to work correctly. However when changing models I would get the switch warning page dialog.  On moving the last switch to the proper position the screen would blank and not wake.  To repeat, I have to switch models a couple of times.  At this time, as soon as I move the brightness slider to the far left the screen blanks and remains blank. I am unable to return the display to on brightness.  To resolve the issue, I remove the SD card and change the display setting using the companion software.  This issue has not presented on my Tx16s MKII.

### Expected Behavior

When the display off brightness is set to zero, I expect the display to return to on brightness when either moving the sticks or a switch.  Mode is set to both.

### Steps To Reproduce

Set display off brightness to zero.  If the issues does not present try switching models or power cycling the radio.

### Version

2.10.2

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

2.10.2

3djc, 2024-10-19:
There is no code change between mk1 and mk2, so it is really unclear to me given what you say where the issue is more likely to be (ie etx issue or broken model)

pfeerick, 2024-10-19:
It would probably help if you upload either a zipped copy of your RADIO and MODELS folders, or a ETX file save from Companion (change the extension to .zip so you can upload it on github) so we can see exactly what settings you are working with.

Astrohog, 2024-10-19:
I made some updates above.  In short while out yesterday flying with the MK II, I was unable to wake the screen.  I took 4 models with me and between flights on the 4th I was unable to wake the screen.  This was not between switching models or clearing the switch prompt.  After cycling the power and changing models several time at home over a few days I thought the MK II did not have this issue but it does.  I attached and etx file with some of the models I was flying yesterday.  Of course in this backup, in the system settings the "backlight-off brightness" is not set to zero in the file.  With the slider all the way to the left is the settings I used yesterday.  I don't feel the issue is model dependent but is rather an EdgeTx issue but I'm just guessing.
[TX16s MKII 2.10.2.zip](https://github.com/user-attachments/files/17445337/TX16s.MKII.2.10.2.zip)