EdgeTX/edgetx #5639: Trainer Settings: Multiplier has no effect

icebreaker-ch, 2024-10-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Multiplier setting in Trainer Settings has no effect. The documentation says:
Multiplier - This value changes the weight of all the sticks together. 
But regardless of the value set for Multiplier, the weight of the sticks is the same.

### Expected Behavior

I expected, to set multiplier for example to 0.5 and the input of the students transmitter weights only by 50%

### Steps To Reproduce

- Setup a trainer configuration by using a Multimodule in the Trainers transmitter.
- Change the value of Multiplier in the Sys->Trainer Settings.
- The weight of all sticks from the student transmitter do not change

### Version

2.10.5

### Transmitter

RadioMaster Zorro

3djc, 2024-10-25:
Unsure what your issue actually is. I did setup a wired trainer, and as you can see, multiplier does indeed affect values

[image]

[image]

icebreaker-ch, 2024-10-25:
And here the same pictures for my wireless Trainer setup:
[image]
[image]

3djc, 2024-10-25:
Ok so a wireless trainer specific issue. Is that multi trainer ?

icebreaker-ch, 2024-10-25:
I can confirm, that it works in a wired trainer setup, but not in my wireless setup.
I used the following combination: Trainer: Radiomaster Zorro 4in1 with a IRangeX IRX4 LITE Multi Protocol Module. Student: Radiomaster T8 Lite.

3djc, 2024-10-26:
In the current code, the value set in this multiplier field is called "PPM_Multiplier" and is applied **only** to DSC (training wire) port. So it does not apply to wireless, SBUS or any other trainer methods. So

1) we can change the UI to only display  this multiplier when DSC input is selected (and use mixer weight if you want to adjust it, which would make sense since you probably don't want the same weight for all control surfaces)

2) we change the code so it is applied for all forms of trainer (at teh risk of breaking existing setup)

icebreaker-ch, 2024-10-26:
I leave it to you, how to solve this issue. Both variants are acceptable to me, because the use case I had (reducing the weight of the students input) can be achieved by setting the weights of the involved channels as well.
To have this in a single setting would have been just a convenience and probably is some kind of abuse of the multiplier factor either.

PR #5649 (2024-11-09): fix: only display PPM multiplier when PPM is used
PPM multiplier is used only on trainer jack input (for history reason when PPM was hardly standardized), but is displayed in all mode. With this change, it is displayed only in master/jack mode

Fixes #5639
closed 2024-11-09 by pfeerick