EdgeTX/edgetx #5640: can not select inverted switch using keys

offer-shmuely, 2024-10-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

selecting a "switch" on a mixer line
the "Invert" button can not be reached with keys (only with touch)
so radio like the X10 can not choose " !SE- "

### Expected Behavior

be able to select "Invert" button with keys as well

### Steps To Reproduce

1. edit a mixer line
2. choose "switch"
3. select "SA-
4. try to navigate to "Invert" with keys

### Version

2.10.4

### Transmitter

FrSky X10 / X10S (ACCST), RadioMaster TX16S / TX16SMK2

### OS Version

NA

### Anything else?

[image]

pfeerick, 2024-10-25:
Looks like this is present in 2.10.5 and nightly also (which is really bugging me as I thought there had been a PR merged to fix  exactly that 🤔)

philmoz, 2024-10-25:
The 'Invert' option is only supposed to be present for radios with touch screens - it is only accessible via touch input.

When this feature was added only radios with factory fitted touch screens had the HARDWARE_TOUCH setting - since then someone has made this the default for ALL color radios regardless of whether they have a touch screen fitted or not. IMO this should not have been changed.

To select inverted mode long press on the ENTER key on the switch control (not in the popup).

3djc, 2024-10-25:
Pretty much like BW radios

philmoz, 2024-10-25:
Is there a reliable way to check at runtime if the touch screen is actually present? (Since we can no longer tell at compile time.)

3djc, 2024-10-25:
I think it is time we create a radio detected struct that list the elements been discovered (touch, serial gimbal, BT,...) that we can query globally (doesn't need to be part of yaml imho)

pfeerick, 2024-10-25:
> When this feature was added only radios with factory fitted touch screens had the HARDWARE_TOUCH setting - since then someone has made this the default for ALL color radios regardless of whether they have a touch screen fitted or not. IMO this should not have been changed.

This is somewhat of a fallacy, as the hardware define can really only dictate whether a radio *might* have touch, not whether it actually does, so the define can only really be used to include or not include stuff that _might_ be relevant for touch. 

A struct of runtime detected features is certainly in need, and also needs to be passed to Companion, so that it can know things about the radio configuration that are not configurable at compile time. Neil was going to write a proposal for that, but I expect other stuff got in the way. 

B&W has the invert option in the popup menu. Am I right in assuming it was not included in the navigation flow on colorlcd was because it was a PITA to implement since it's not one of the group buttons? Just asking, as it would have made more UX sense for it to work.

philmoz, 2024-10-25:
This was discussed at length in #4104 and it was decided that 'Invert' should only be visible for radios with a touch screen.

philmoz, 2024-11-14:
In PR #5663, I have changed the Switch and Source popups to allow the long press of the ENTER key to change the 'Invert' state when the popup menu is open (as well as when it is closed).

PR #5663 (2024-11-23): fix(color): cannot change 'Invert' state on radios with no touch screen
Allow long press of ENTER to change 'Invert' state in switch and source popup menus when menu is open.

Fixes #5640
closed 2024-11-23 by pfeerick